BruceDevices/firmware #897: Device only powers on when connected via USB (LILYGO T-Embed CC1101)

nys87, 2025-03-09:
### Issue: Device only powers on when connected via USB (LILYGO T-Embed CC1101)

**Device:** LILYGO T-Embed CC1101  
**Firmware:** BRUCE (flashed via [https://bruce.computer/flasher](https://bruce.computer/flasher))

---

### Problem Description:
After flashing my **LILYGO T-Embed CC1101** via [bruce.computer/flasher](https://bruce.computer/flasher), the device **only powers on when connected via USB Type-C cable** to a computer. It does not respond to the power button press when operating independently (without USB).

### Steps Taken:
- Reflashed the firmware multiple times; the issue persists.
- Confirmed the physical functionality of the power button. It works as expected once the device is powered via USB.
- The button seems ineffective only for initial device startup when not connected via USB.

### Expected Behavior:
The device should power on independently using the physical power button without requiring a USB connection.

---

Any guidance or suggestions to resolve this issue would be greatly appreciated.

stevedee78, 2025-03-09:
The same problem also exists with the Yellow Cheap Display

peppemas1, 2025-03-09:
same for me, but now battery dont drain

Mrod31, 2025-03-09:
Same problem with 1.9 and 1.9.1 and with the launcher

bmorcelli, 2025-03-09:
It is not a problem.. it is a product design..

If you use **Config>Deepsleep** you will be able to wake it up hitting the upper button, but in this mode CC1101 and NRF24 keeps draining battery..

If you use **Config>Turn Off**, you need plug the USB cable or press the power button on the board (take the battery and press it), im this case theres no battery draining.

This is a complaint you need to address to Lilygo...

closed 2025-03-27 by IncursioHack

mmn149, 2025-09-01:
It seems, that the battery drains anyway even after "turn off" ... Haven't used it for am month and the voltage is substantially lowered ...

sffmail80-dev, 2025-12-09:
> It is not a problem.. it is a product design..
>
> If you use **Config>Deepsleep** you will be able to wake it up hitting the upper button, but in this mode CC1101 and NRF24 keeps draining battery.. [...]

why not design so it can be turned on from buttons, or at least from an addon button in the accesible pins?