BruceDevices/firmware #904: Only 3,44 GB of entire SD-Card is

mileshkin, 2025-03-10:
**Describe the bug:**
Bruce, starting with version 1.9 and higher, it does not show the entire volume of the SD card, but only 3.44 GB if viewed through WebUI (also in Launcher of latest versions)
In versions 1.8, 1.8.2, with a 64 GB card installed, it showed its real volume, i.e. 59.44 GB (also in older versions of Launcher)

**Steps To Reproduce:**
Go to WebUI:

[image]

- Device used: 
T-Embed CC1101

**Expected behavior:**
With a 64 GB card installed, it showed its real volume, i.e. 59.44 GB

[image]

bmorcelli, 2025-03-10:
I believe I fixed it in [here](https://github.com/pr3y/Bruce/commit/5021c1bebc00bd87a6406541bfd39c20c26ef33d) .. can you test the last Beta?

mileshkin, 2025-03-11:
[image]
Yyyep! Everything is fine now! Thank you very much!

closed 2025-03-11 by mileshkin