BruceDevices/firmware #905: good feature request

tommyfart, 2025-03-10:
In the BLE section can you add AppleJuice if you know what that is, it would be cool:

[image]

basically it shows a bunch of stuff that apple makes like AirPods, AirPods Max and more, and you can advertise the AirPods and stuff, the reason im requesting this is because iOS Spam does not work, here is what applejuice kinda looks like:

[image]

MikeOxlong83, 2025-03-22:
What FW is that which has this AppleJuice you speak of?

tommyfart, 2025-03-23:
> What FW is that which has this AppleJuice you speak of?

nemo

IncursioHack, 2025-03-27:
There have been several changes made to BLE in the last few weeks, please test it in BETA at https://bruce.computer/.

closed 2025-05-26 by bmorcelli