BruceDevices/firmware #909: I made tetris for bruce

Gosheto1234, 2025-03-12:
**Is your feature request related to a problem? Please describe.**
its just a tetris game located in others 

**Describe the solution you'd like**

**Describe alternatives you've considered**
it works on m5stickplus2 , but i think it should be dynamic and work on the other devices , i havent tested that but i hope it works , the controls : 
For Touchscreen Devices:

Open Menu: Tap the top-right corner (within a 50×50 pixel area).
Move Left: Tap the left third of the screen.
Move Right: Tap the right third of the screen.
Rotate Piece: Tap the center of the screen (upper half).
Drop Piece: Tap the center of the screen (lower half).
For Devices with 3 Buttons:

Open Menu: Press both the “Next” and “Prev” buttons at the same time.
Move Right: Press the “Next” button.
Move Left: Press the “Prev” button.
Rotate Piece: Press the “Sel” (Select) button.
For Devices with 5 Buttons:

Rotate Piece: Press the “Up” button.
Drop Piece: Press the “Down” button.
Move Right: Press the “Next” button.
Move Left: Press the “Prev” button.
Open Menu: Press the “Sel” (Select) button.
For Devices with an Encoder:

Rotate Piece: Press the “Sel” button.
Move Right: Turn the encoder or press the “Next” button.
Move Left: Turn the encoder the other way or press the “Prev” button.
Drop Piece: Press the “Esc” button.
For Keyboard Input:

Move Left: Type “LEFT”
Move Right: Type “RIGHT”
Rotate Piece: Type “UP”
Drop Piece: Type “DOWN”
Open Menu: Type “MENU”

**Additional context**
how do i request my files to be pulled into beta code ? ( the tetris in very beta so feedback will be good , thanks in advance if this feature gets added )

Huzzla101, 2025-03-13:
Its on FIRE.... ! If its allready intact then I cant wait to try this on
the LILYGO T EMBED CC1101 that have the sickest color screen... So when can
we expect the next game.. A RPG or something ..With Dragons and all
that.... :p And also it would be sick to have a RF and or bluetooth  game
that includes maybe a little story line and some cool features but thats in
the future tho.. But with the Magical appearance from Mr Pirata and his
Coders i see it all possible with time... :p

MikeOxlong83, 2025-03-24:
Where do I find this Tetris? I'm a Tetris-holic and I need to try this!!!!!

Gosheto1234, 2025-03-25:
> Where do I find this Tetris? I'm a Tetris-holic and I need to try this!!!!!

I have the tetris on my folder , like all files , I don't know if it runs on every device but in theory it should , I can send you the files if you can flash it onto m5stickplus2 from visual code studio

MikeOxlong83, 2025-04-07:
I use a CYD, not an m5stick but yes please. share the files

Gosheto1234, 2025-04-07:
> I use a CYD, not an m5stick but yes please. share the files

sorry man , i really lost the files , i cant find them

closed 2025-04-07 by Gosheto1234

MikeOxlong83, 2025-04-09:
RUFKM? *facepalm*

Don't you have it on YOUR DEVICE? Can't you just rip it from there?

Gosheto1234, 2025-04-09:
> RUFKM? _facepalm_
> 
> Don't you have it on YOUR DEVICE? Can't you just rip it from there?

It was on my device just for test before time , but I had to switch to beta version , basically I don't have it