BruceDevices/firmware #910: Add support for ESPBOY

coppaxamsukk, 2025-03-12:
Espboy is a diy esp8266 based project that's similar to cardputer but available from long ago. 

Can you port the awesome BRUCE for espboy?
https://www.espboy.com/

[image]

bmorcelli, 2025-03-14:
idk.. I don't have any ESP8566 board, and many libraries are made specifically to ES32, so most probably it won't be officially ported

coppaxamsukk, 2025-03-14:
> idk.. I don't have any ESP8566 board, and many libraries are made specifically to ES32, so most probably it won't be officially ported

I'm sure bruce will work, some features of bruce already included in espboy.

 like GSM phone
GPS navigator
Meteostation
LORA messenger
RFID/NFC reader/repeater
Sub1Ghz inspector
Sub1Ghz scanner
MOD/S3M/XM/IT player
AY38910/12 Yamaha chip player
Web radio
FM radio
Walkie-talkie
Meccano rover
Flower watering
Ice fishing automation
Volt/Amper meter
CW radio beacon
RSS reader
Digital ruler
Pinger
Digital thermometer
Heart rate meter
WiFi analizer
WiFi sniffer
WiFi packet monitor
WiFi pinger
WiFi deauther
IR blaster
Geiger counter
I2C scanner
MP3 player

MikeOxlong83, 2025-03-24:
Wow, quite the device! First I've heard of it too!

coppaxamsukk, 2025-03-26:
> Wow, quite the device! First I've heard of it too!

Yep, check it out.

coppaxamsukk, 2025-03-26:
> idk.. I don't have any ESP8566 board, and many libraries are made specifically to ES32, so most probably it won't be officially ported

Esp32 based esp32boy is coming soon. Hope you then port it❤️

closed 2025-05-26 by bmorcelli