BruceDevices/firmware #911: File sharing lacks of packet validation.

zcaliptium, 2025-03-13:
## Problem
Bruce firmware has file sharing feature that based on ESPNOW protocol. The main problem ESPNOW can be used by various devices around for different purposes. And on current implementation there are no data packet validation.

So we can get non-Bruce ESPNOW packet and it will mess everything up.

## Solution
Add some Magic Number / Identifier into beginning of data packet to understand that we get FileMessage from Bruce device and nothing else.

```cpp
#define BRUCE_ESPNOW_ID "BRUCENET"

    struct FileMessage {
        char magic[8]; // <- put this to be sure that its bruce file sharing
        String filename;
        String filepath;
        size_t totalBytes;
        size_t bytesSent;
        char data[PAGE_BYTES];
        size_t dataSize;
        bool done;
        bool ping;
        bool pong;

        // Constructor to initialize defaults
        FileMessage()
            : totalBytes(0), bytesSent(0), dataSize(0), done(false), ping(false), pong(false) {
            memcpy(&magic[0], BRUCE_ESPNOW_ID, 8); // write it into memory
          }
        }
    };
```

Then put safety check for magic within Receive Callback.
```cpp
void onDataRecv(const uint8_t* mac, const uint8_t* incomingData, int len) {
    if (strcmp((const char *)incomingData, BRUCE_ESPNOW_ID) != 0) {
        return; // ignore non-Bruce packets
    }
```

zcaliptium, 2025-03-19:
Currently I am working on a big Refactor/Pull Request for ESPNOW subsystem on own Fork. I will send it as it finish it. I already fixed that issue there as well as totally reworked data exchange protocol.

rennancockles, 2025-03-19:
Just make sure your branch is up to date with our main branch as espnow is being used for other purposes now too

closed 2025-05-26 by bmorcelli