BruceDevices/firmware #912: Full Bruce Funktions for T-deck

tgb55, 2025-03-13:
Hello, I'm wondering if it's possible to add all Funktions from Bruce to the t-deck. 

I can deal with hardware but my skills for editing software is bad. So I want to solder nrf24 and c1101 to the T-deck. Using the pins from esp32 directly. 

Maybe anyone can help me with that?

xukashi, 2025-03-14:
I also wish that the T-deck gets a little more attention. The grove connector makes it possible to use many modules from m5stack. It would also be nice that the touchscreen works. So I'm currently using another firmware where it all works.

tgb55, 2025-03-14:
> I also wish that the T-deck gets a little more attention. The grove connector makes it possible to use many modules from m5stack. It would also be nice that the touchscreen works. So I'm currently using another firmware where it all works.

Can you tell me the name of the firmware you are using? The only other firmware I know is "acid drop"

xukashi, 2025-03-14:
> Can you tell me the name of the firmware you are using? The only other firmware I know is "acid drop"

ESP32 Marauder https://github.com/jstockdale/ESP32Marauder-T-Deck/tree/t-embed

tgb55, 2025-03-14:
> ESP32 Marauder https://github.com/jstockdale/ESP32Marauder-T-Deck/tree/t-embed

Ahh okay. It was to obvious to think about it^^ only tried it on basic esp32 before, I will test it when I got my T-deck plus, thx!

bmorcelli, 2025-03-19:
I enabled all functions on Bruce.. some of them (cc1101, nrf24) youll need to setup the pins on bruce.conf

available on the BetaRelease for testing..

tgb55, 2025-03-19:
Hell yeah thx, that makes this project more flexible! I mean, an blackberry looking aio hacking device is my dream^^

xukashi, 2025-03-19:
Unfortunately, the touchscreen and GPS don't work. I found the following information about the GPS:

UBlox MIA-M10Q GNSS Module
UBlox Module using baud rate 38400
GPIO44 for GPS RX
GPIO43 for GPS TX

bmorcelli, 2025-03-19:
This is not the Plus, ryt?

xukashi, 2025-03-19:
i have the Plus.

closed 2025-05-26 by bmorcelli