BruceDevices/firmware #913: Bruce fw isn't responding to button inputs on esp32-s3 devkit .

Z-SHOCK-E, 2025-03-13:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-s3-devkitc-1-psram]
board_build.arduino.memory_type = qio_opi  ; NEEDED FOR 8MB PSRAM (N16R8)
;board_build.partitions = custom_16Mb.csv
;board_upload.flash_size = 16MB
;monitor_speed = 115200
;board_build.partitions = custom_16Mb.csv
;board_build.f_flash = 40000000L
;board_upload.flash_size = 16MB
;board_upload.maximum_size = 16777216
;board_build.f_flash = 40000000L
;board_build.flash_mode = qio
extends                 = env:esp32-s3-devkitc-1

[env:esp32-s3-devkitc-1]
board = ESP-General
build_src_filter =${env.build_src_filter} +<../boards/ESP-General>
build_flags =
	${env.build_flags}
	-Iboards/ESP-General
	-DUSB_as_HID=1
	; grove pins
	; defaults from https://github.com/espressif/arduino-esp32/blob/master/variants/esp32s3/pins_arduino.h
	-DGROVE_SDA=8  ; default RF TX pin
	-DGROVE_SCL=46  ; default IR/RF RX pin
	;-DALLOW_ALL_GPIO_FOR_IR_RF=1 ; Set this option to make use of all GPIOs, from 1 to 44 to be chosen, except TFT and SD pins

	; ir led pin
	-DIR_TX_PINS='{{"M5 IR Mod", GROVE_SDA}, {"Grove W", GROVE_SCL}, {"Grove Y", GROVE_SDA}}'
	-DIR_RX_PINS='{{"M5 IR Mod", GROVE_SCL}, {"Grove W", GROVE_SCL}, {"Grove Y", GROVE_SDA}}'
	-DLED=40
	-DLED_ON=HIGH
	-DLED_OFF=LOW

	;Radio Frequency (one pin modules) pin setting
	-DRF_TX_PINS='{{"M5 RF433T", GROVE_SDA}, {"Grove W", GROVE_SCL}, {"Grove Y", GROVE_SDA}}'
	-DRF_RX_PINS='{{"M5 RF433R", GROVE_SCL}, {"Grove W", GROVE_SCL}, {"Grove Y", GROVE_SDA}}'

	; sd card pins
	; suggested https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc
	-DSDCARD_CS=-1
	-DSDCARD_SCK=-1
	-DSDCARD_MISO=-1
	-DSDCARD_MOSI=-1

	; tft vars
	-DROTATION=1
	-DBACKLIGHT=-1  ; tft backlight pin
	-DMINBRIGHT=160  ; unused?
	-DSMOOTH_FONT=1
	-DTFT_DISPON=0x29
	-DTFT_DISPOFF=0x28
	-DTFT_CS=-1
	-DTFT_DC=-1
	-DTFT_RST=-1
	-DTOUCH_CS=-1
	-DTFT_MOSI=-1
	-DTFT_SCLK=-1
	-DTFT_BL=-1
	; text sizes
	-DFP=1
	-DFM=2
	-DFG=3
	; ui control buttons
    
    -DHAS_BTN=1
	-DSEL_BTN=11
	-DUP_BTN=10  ; also work as ESC
	-DDW_BTN=9  ; also work as NEXT
	-DBTN_ACT=LOW
	-DBTN_ALIAS='"OK"'

	;FM Radio
	;-DFM_SI4713=1 ;Uncomment to activate FM Radio using Adafruit Si4713
	-DFM_RSTPIN=12
	;Microphone
	;-DMIC_SPM1423=1 ; uncomment to enable Applicable for SPM1423 device
	;-DPIN_CLK=-1
	;-DI2S_SCLK_PIN=-1
	;-DI2S_DATA_PIN=-1
	;-DPIN_DATA=-1

	;CC1101 SPI connection pins
	; best connection pins for higher speed https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux
	-DUSE_CC1101_VIA_SPI
	-DCC1101_GDO0_PIN=9
	-DCC1101_SS_PIN=10
	-DCC1101_MOSI_PIN=11
	-DCC1101_SCK_PIN=12
	-DCC1101_MISO_PIN=13
	;-DCC1101_GDO2_PIN=14  ; optional

	; connections are the same as CC1101
	;-DUSE_NRF24_VIA_SPI
	;-DNRF24_CE_PIN=6
	;-DNRF24_SS_PIN=7  ; chip select
	;-DNRF24_MOSI_PIN=11
	;-DNRF24_SCK_PIN=12
	;-DNRF24_MISO_PIN=13
	-DUSE_HSPI_PORT=1
	-DHAS_SCREEN=1
	-DSPI_SCK_PIN=12
	-DSPI_MOSI_PIN=11
	-DSPI_MISO_PIN=13
	-DSPI_SS_PIN=10

lib_deps =
	${env.lib_deps}`

^ ESP-General.ini

I managed to get the 1.9" TFT display working with bruce and I connected some push buttons from GND(Common) to the defined button pins. After connecting none of them worked.
I tried using INPUT, INPUT_PULLUP and INPUT_PULLDOWN, but none of them worked. It booeted up skipped the boot animation and started doing stuff on it's own.

interface.cpp

'#include "interface.h"
#include "core/powerSave.h"

#include <driver/adc.h>
#include <esp_adc_cal.h>
#include <soc/soc_caps.h>
#include <soc/adc_channel.h>

/***************************************************************************************
** Function name: _setup_gpio()
** Location: main.cpp
** Description:   initial setup for the device
***************************************************************************************/
void _setup_gpio() { 
    pinMode(UP_BTN, INPUT);   // Sets the power btn as an INPUT
    pinMode(SEL_BTN, INPUT);
    pinMode(DW_BTN, INPUT);
}
/***************************************************************************************
** Function name: getBattery()
** location: display.cpp
** Description:   Delivers the battery value from 1-100
***************************************************************************************/
int getBattery() { return 0; }

/*********************************************************************
** Function: setBrightness
** location: settings.cpp
** set brightness value
**********************************************************************/
void _setBrightness(uint8_t brightval) { }

/*********************************************************************
** Function: InputHandler
** Handles the variables PrevPress, NextPress, SelPress, AnyKeyPress and EscPress
**********************************************************************/
void InputHandler(void) { 
    checkPowerSaveTime();
    PrevPress    = false;
    NextPress    = false;
    SelPress     = false;
    AnyKeyPress  = false;
    EscPress     = false;

    if(digitalRead(SEL_BTN)==BTN_ACT || digitalRead(UP_BTN)==BTN_ACT || digitalRead(DW_BTN)==BTN_ACT) {
        if(!wakeUpScreen()) AnyKeyPress = true;
        else goto END;
    }    
    if(digitalRead(UP_BTN)==BTN_ACT) {
        UpPress = true;
    }
    if(digitalRead(DW_BTN)==BTN_ACT) {
        DownPress = true;
        EscPress = true;
    }
    if(digitalRead(SEL_BTN)==BTN_ACT) {
        SelPress = true;
    }
    END:
    if(AnyKeyPress) {
      long tmp=millis();
      while((millis()-tmp)<200 && (digitalRead(SEL_BTN)==BTN_ACT || digitalRead(UP_BTN)==BTN_ACT || digitalRead(DW_BTN)==BTN_ACT));
    }
}

/*********************************************************************
** Function: powerOff
** location: mykeyboard.cpp
** Turns off the device (or try to)
**********************************************************************/
void powerOff() {
    esp_sleep_enable_ext0_wakeup((gpio_num_t)UP_BTN,HIGH); 
    esp_deep_sleep_start();

 }

/*********************************************************************
** Function: checkReboot
** location: mykeyboard.cpp
** Btn logic to tornoff the device (name is odd btw)
**********************************************************************/
void checkReboot() { 
}
'

Z-SHOCK-E, 2025-03-13:
It finall worked ( kind of ). The select button works buto the up and down ones only work inside the WiFI subMenu. I can't test the other menu's bcuz the up and down keys aren't working on the main UI page

bmorcelli, 2025-03-13:
They aren't supposed to work on main menu..
Only in submenus and file menus

Z-SHOCK-E, 2025-03-13:
So, how can I navigate between the main menus. By the main menus I mean WiFi, BLE, FILES.

bmorcelli, 2025-03-13:
if(digitalRead(SEL_BTN)==BTN_ACT || digitalRead(UP_BTN)==BTN_ACT || digitalRead(DW_BTN)==BTN_ACT) {
    if(!wakeUpScreen()) AnyKeyPress = true;
    else goto END;
}    
if(digitalRead(UP_BTN)==BTN_ACT) {
    NextPress = true;
}
if(digitalRead(DW_BTN)==BTN_ACT) {
    PrevPress = true;
    EscPress = true;
}
if(digitalRead(SEL_BTN)==BTN_ACT) {
    SelPress = true;
}
END:
if(AnyKeyPress) {
  long tmp=millis();
  while((millis()-tmp)<200 && (digitalRead(SEL_BTN)==BTN_ACT || digitalRead(UP_BTN)==BTN_ACT || digitalRead(DW_BTN)==BTN_ACT));
}

Z-SHOCK-E, 2025-03-13:
Thanks a LOT. It works.

Z-SHOCK-E, 2025-03-13:
The buttons will work everywhere except for the on_screen keyboard. Why won't work

bmorcelli, 2025-03-13:
You mean the touchscreen?

Z-SHOCK-E, 2025-03-13:
No no. The same keyboard that is on the m5stick2. Plus how to make the devkit power off when holding ESC button.

bmorcelli, 2025-03-13:
You need add
 ```-DHAS_3_BUTTONS=1```
Into your .ini environment

Z-SHOCK-E, 2025-03-13:
and remove the -DHAS_BUTTON=1  ?

Z-SHOCK-E, 2025-03-13:
Sure, but where is it ?

bmorcelli, 2025-03-14:
baby shark, tururururu, baby tururururu, baby shark, **tururururu.cpp**

closed 2025-03-14 by bmorcelli

sivabala21, 2025-07-22:
> Sure, but where is it ? I can't find a file named 'megalodon.cpp '

ZshockE I want contact you,
any mail or discord id please.