BruceDevices/firmware #917: (M5 stickCplus2) screen sometimes goes inverted randomly and fixes after reboot

Fireblade497, 2025-03-13:
**Describe the bug:**

**Steps To Reproduce:**

- Device used: 

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

bmorcelli, 2025-03-13:
This is odd.. never seen it.. can you describe when it happens or a way for us to reproduce it?

Fireblade497, 2025-03-13:
> This is odd.. never seen it.. can you describe when it happens or a way for us to reproduce it?

I can’t reproduce it because it happens at random

GabriWar, 2025-03-18:
what do you mean by inverted? the screen flips? i think ive seen it happen maybe once on the tembed, maybe it has something to do with the bruce.conf?

bmorcelli, 2025-03-18:
Probably something is "overflowing" into some tft address and making tft act weird..

I've seen that there are some issues with the InputHandler task, maybe it is causing this problem

Fireblade497, 2025-03-18:
> what do you mean by inverted? the screen flips? i think ive seen it happen maybe once on the tembed, maybe it has something to do with the bruce.conf?

The colors invert

DDLS10, 2025-03-20:
do you use an sd card and do you share it with other bruce devices

Fireblade497, 2025-03-20:
> do you use an sd card and do you share it with other bruce devices

No