BruceDevices/firmware #918: StickC Plus2 cc1101

Michaeli47, 2025-03-13:
i just bought my first stick the rf was working when i tried the first time but when i tried again it won't catch any frequency's when i configure and attempted to put it in shared spi it just crashes will only run on legacy without it freezing but wont get any frequency's

bmorcelli, 2025-03-13:
If using scan/copy to scan the frequencies, change the Threshold to -70 or -80, so you can get something.. then you narrow the field and try higher values (-60).. in this hopping strategy (range scan), the range is pretty limited due to the speed it runs to detect something.. so whenever os possible, use a fixed frequency and you'll get better results.

Frequencies different from your CC1101 specs will get lower RSSI due to the antenna circuit, and it means short ranges.

Once you detect the frequency, start using fixed freqs

Michaeli47, 2025-03-13:
It was the female part of the connector I tightened it down now its perfect

closed 2025-03-21 by IncursioHack