BruceDevices/firmware #921: ✨PROPOSAL✨: Would it be possible to make a port for T watch S3 plus? (it has ESP32-S3)

Cha350, 2025-03-14:
I cant help but wonder if it would be possible to make a port of bruce for T watch S3 plus? Im wondering because if this is possible, it would be good as this version of the T watch has 940mAh battery instead of 470mAh battery. This also has added gps but other than those two things, i think its pretty much the same? here are some info pictures.

[image]
[image]
[image]

Here is the link to the page: https://lilygo.cc/products/t-watch-s3-plus?pr_prod_strat=e5_desc&pr_rec_id=a87af1d05&pr_rec_pid=8173536477365&pr_ref_pid=7454245224629&pr_seq=uniform

bmorcelli, 2025-03-14:
the Version for T-Watch S3 is compatible with Plus, GPS is already configured for it

give it a try

Cha350, 2025-03-14:
> the Version for T-Watch S3 is compatible with Plus, GPS is already configured for it
> 
> give it a try

Oh ok! I didnt know that the T watch S3 port could be compatable with the T watch S3 plus. Thank you for telling me know 😊

closed 2025-03-18 by bmorcelli

Safaryokla, 2025-03-19:
O Bruce não tem suporte para o T DISPLAY S3 pro dá Lilygo?

bmorcelli, 2025-03-19:
No, not yet