BruceDevices/firmware #925: no exit button on CYD rfid reading option

Gosheto1234, 2025-03-15:
**Describe the bug:
when using CYD with 2 usbs using  PN532 module , after reading a nfc chip in select modes there is no exit at all , can you add exit function ? , also sometimes the select buttont doesnt open the changing mode menu in the Rfid function
**Steps To Reproduce:**

- Device used: CYD-2432S028 2USBs

**Expected behavior:**

i expect to have exit option , and opening the menu 
**Actual behavior:**

**Additional context / logs:**

bmorcelli, 2025-03-15:
Try touching the top left corner

```
                    Touch area Map
            ________________________________ 0
            |   Esc   |   UP    |         |
            |_________|_________|         |_> third_y
            |         |   Sel   |         |
            |         |_________|  Next   |_> third_y*2
            |  Prev   |  Down   |         |
            |_________|_________|_________|_> third_y*3
            |__Prev___|___Sel___|__Next___| 20 pixel touch area where the touchFooter is drawn
```

Gosheto1234, 2025-03-15:
thanks , it worked

closed 2025-03-15 by Gosheto1234