BruceDevices/firmware #926: Reboot after deaut target

RoToSpLaSh, 2025-03-15:
**Describe the bug:**
Reboot after deauth target 

**Steps To Reproduce:**

- Device used: CardPuter 
Connect to my wifi and scan hosting, after i try to deauth target and this cause Reboot 

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

IncursioHack, 2025-03-21:
First of all, I would like to thank you for being one of the few who, in addition to the information, also brought the Serial Log and test data. Congratulations on the care in bringing this data.

I know that we have had some recent changes in Bruce. Would it be possible for you to test the same function under the same conditions with the BETA RELEASE version?

bmorcelli, 2025-04-03:
Fixed in Beta

RoToSpLaSh, 2025-04-16:
The bug is non fixed.