BruceDevices/firmware #928: The flasher did not install bruce on T-Embed CC1101

Zull1n0, 2025-03-16:
When I try to install Bruce on my T-Embed CC1101, Pc tell me: "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL."
How I can resolve?
Thanks

Gosheto1234, 2025-03-16:
try reading here https://github.com/pr3y/Bruce/issues/846#issuecomment-2683226395

Huzzla101, 2025-03-16:
BOOT button is simply the button in the middle of the Great Wheel.... Turn
it off without the usb plugged then hold that button when you enter the usb
then it wont turn on but goes to BOOT MODUS and  if needed check that your
pc gives it a com port... It is actually that easy =)

(if your pc dont give it a com port you can also deactivate then activate
it, the usb 3.10 extensible host controller is my name... Yours most likely
is about the same name  with processor name in front AMD etc in your
device manager/USB CONTROLLERS,  then it should work , worst case
restart...And make sure you using a usb with data transfer not just charger
)

Zull1n0, 2025-03-18:
In my T-Embed CC1101 I've installed Capibara.

I've tried all suggestion in the other post with no results.

Thanks
P.s. please don't close this post to let me tell how the problem evolving

Zull1n0, 2025-03-19:
> try reading here [#846 (comment)](https://github.com/pr3y/Bruce/issues/846#issuecomment-2683226395)

Great!!!! It now work!!! Thanks and Best Regards!!!