BruceDevices/firmware #929: Smoochie Board flickering

Melman5123, 2025-03-16:
When smoochie-board.bin is uploaded to esp32s3 it causes the display to flicker and unresponsive to input, it moves between menus by itself. I have wired according to smoochie-board.h and got this result

https://github.com/user-attachments/assets/588cdf69-a883-44b0-8b98-6a85d9049f4f

bmorcelli, 2025-03-18:
Your computer is sending invalid serial commands.. sometimes it happens to me with other devices 

And the auto navigation probably is that you need to pull up the button gpio, and drive the button to gnd when pressed

Melman5123, 2025-03-18:
> Your computer is sending invalid serial commands.. sometimes it happens to me with other devices 
> 
> And the auto navigation probably is that you need to pull up the button gpio, and drive the button to gnd when pressed

Perfect thanks for the help, got it to work using pull up 10k resistors. However none of the devices I connect either over spi or i2c can be detected, I have a pn532, nrf24 and cc1101. When connected pn532 over i2c, the rest over spi, it says module not detected or whatever. Not sure what to do, I'm following the smoothie board pinouts, can you help please?

Shadow88633, 2025-03-22:
Can i ask you about your pinout, mine doesnt work lol

Melman5123, 2025-03-26:
> Can i ask you about your pinout, mine doesnt work lol

Ye sure go ahead sorry I didn't see this comment as I have mine working and wasn't checking

closed 2025-03-28 by bmorcelli

azizbekrakhimjonov, 2025-06-14:
nice