BruceDevices/firmware #932: Pre-released FW - m5stick plus2 - No NRF24 found

Michaelicious, 2025-03-17:
**Describe the bug:**
Flashing pre released FW caused the cplus2 to not find nrf24 module.

**Steps To Reproduce:**
1. Flash pre realesed fw
2. Connect NRF24 chip
3. You will not find it

- Device used: 
M5Stack CPLUS2

**Expected behavior:**
nrf24 found

**Actual behavior:**
"No nrf24 found" message

**Additional context / logs:**

IncursioHack, 2025-03-21:
I just re-recorded my StickC and had no problems, could you send detailed images of your connection? We have seen several reports in the Discord group about this topic for months in a row and in all of them it was either the wrong cable connection or the module was defective.

IncursioHack, 2025-03-26:
We need a lot of information to analyze your case, to report a bug we leave a manual described of what is needed so that we can verify the problem. This manual is inside Bruce's discord channel in the chat #bug-report-rules. After attaching the information, we will be able to analyze what it is about.

Edersonrocha22, 2025-04-07:
HELP ME

https://github.com/user-attachments/assets/e299079c-5e6f-428d-9f20-061506cc3ecd