BruceDevices/firmware #934: Ibutton t ember

loni1488, 2025-03-17:
Hello, I would like to know how to open doors with ibutton using t-embet cc1101, I can’t find a diagram, I have an empty hat without anything, how can I somehow make a module out of it, I also heard that A resistor will be needed, is that true?

bmorcelli, 2025-03-17:
It wasn't tested with T-embed yet, and I saw in a quick look that it is set only for StickCPlus devices, as it is hardcoded for gpio 0 (zero).. so we need to add an option to change this value somewhere..

iButton can do read and write into a button, but our implementations isn't emulating yet..

So, I'm counting on the community to develop the emulation, as I don't have anything to test and develop myself

bmorcelli, 2025-03-18:
I just send a new PR where you can choose the pin to use thi ibutton thing.. so you can setup to gpio 43 or 44 and try to wire the ibutton there.

Will be available soon enough

closed 2025-03-18 by bmorcelli

bmorcelli, 2025-03-19:
> https://github.com/user-attachments/assets/af36af26-7241-4ad1-ac4c-8f5b33f3245e So this video is a fake? And yes, it is correct that you need to connect one pin of the shield vcc And the other pin is a hat on the pin date ?

I won't say it is fake, because probably he built the project at home with the right button settings..

Now in the beta version you can try it.. connect vcc, gnd and data pin to the QWIIK connector correspondent to gpio 43 or 44 and give it a try

bmorcelli, 2025-03-19:
Other thing important is that you can read and write to these iButtons, not emulate.. so if he is emulating, it is probably fake