BruceDevices/firmware #940: JC2432W328C Build Errors "libnet80211.a"

dankcincy, 2025-03-18:
I'm new to all this and am not sure what to do here. 
Any help would be greatly appreciated

Processing CYD-2432W328C (board: CYD-2432S028; platform: espressif32; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing h2zero/NimBLE-Arduino
Library Manager: Installing h2zero/NimBLE-Arduino
Library Manager: Installing h2zero/NimBLE-Arduino
Verbose mode can be enabled via -v, --verbose option
[GZIP & EMBED INTO HEADER] - Nothing to process.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/CYD-2432S028.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.17
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
---------------------------------------------------------------------------------------------------

I try to build this and am getting this error,

c:/users/g90 precision/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lnet80211
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\CYD-2432W328C\firmware.elf] Error 1
====================================================================================== [FAILED] Took 86.95 seconds ======================================================================================
Environment    Status    Duration
-------------  --------  ------------
CYD-2432W328C  FAILED    00:01:26.948
================================================================================= 1 failed, 0 succeeded in 00:01:26.948 =================================================================================
 *  The terminal process "C:\Users\G90 Precision\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'CYD-2432W328C'" terminated with exit code: 1. 
---------------------------------------------------------------------------------------------------------------------------

bmorcelli, 2025-03-18:
rename the file:
`C:\Users\G90 Precision.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\lib\libnet80211.a.patched`

to

`C:\Users\G90 Precision.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\lib\libnet80211.a`

if you will build for ESP32S3 too, you will need to do it for the esp32s3 folder too
`C:\Users\G90 Precision.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s3\lib\libnet80211.a.patched`

closed 2025-03-18 by bmorcelli

dankcincy, 2025-03-18:
This is what I have,

C:\Users\G90 Precision\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\lib\libnet80211.a.old
C:\Users\G90 Precision\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32c3\lib\libnet80211.a
C:\Users\G90 Precision\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s2\lib\libnet80211.a
C:\Users\G90 Precision\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s3\lib\libnet80211.a

bmorcelli, 2025-03-19:
Delete the whole packages and platform folders fro platformio and open vscode..

Ot will download all framework files again

And comment out all these lines:
https://github.com/pr3y/Bruce/blob/main/patch.py#L21-L43