BruceDevices/firmware #945: I'm repeat the question after closing similar

Zull1n0, 2025-03-18:
In my T-Embed CC1101 I've installed Capibara.
Now when I try to install Bruce on my T-Embed CC1101, Pc tell me: "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL."
How I can resolve?
I've tried all suggestion in the other post with no results. 

Thanks
P.s. please don't close this post to let me tell how the problem evolving

mileshkin, 2025-03-19:
Enter the boot mode pressing and holding center button then press RST button on the mainboard of T-Embed then release them both. Unit must enter BOOT and you would flash it normally on any Bruce/Launcher/Capibara/Anything.. you must see blank screen in BOOT mode (else it is not BOOT)
If not then you must check RST button to work.

Zull1n0, 2025-03-19:
Thanks but it did not work. When I insert the USB-C in T-Embed, it start with Capibara and the web flasher return the same message.

mileshkin, 2025-03-19:
> Thanks but it did not work. When I insert the USB-C in T-Embed, it start with Capibara and the web flasher return the same message.

Try to do reset to BOOT with USB inserted.

Zull1n0, 2025-03-19:
Did not word. If USB is inserted, T-Embed did start with Capibara ever.

Zull1n0, 2025-03-19:
I resolved with https://github.com/pr3y/Bruce/issues/928#issuecomment-2727368524
thanks to all!!!

bmorcelli, 2025-03-19:
Take the back cover
Connect your t-embed into the USB-C
**HOLD** the selection button, the one in the middle of the big wheel
Poke the RST button on the board with a tooth pick

**Now screen should be black, in download mode**

Release the Select button

Install [Launcher](https://bmorcelli.github.io/M5Stick-Launcher)

Poke the RST button once again, to restart the device

[image]

After that, download the binaries and save on your SDCard and install from there.

Bruce binaries are here https://github.com/pr3y/Bruce/tree/WebPage

In the LastRelease or BetaRelease folder

Idk anything about capybara, so I can't help you with that

closed 2025-03-19 by bmorcelli

Zull1n0, 2025-03-19:
Thanks!!! Best Regards!!!