BruceDevices/firmware #948: ❗️ BADUSB Feature Extension ❗️

mamadavi, 2025-03-20:
**Hi, I would like to make a very important request to you. I would like to ask you to expand the functionality of BADUSB and put more emphasis on this. I know that there are a couple of bugs and shortcomings in this feature, so I ask you to fix it and make the menu more convenient. I think it's worth taking BADUSB more seriously and allocating separate slots for attacks like BADUSB, BADBLE with some standard scripts like rickroll or something else. It would also be nice to support other modules for attacks of this type. I want you to add arduino support and make the bruce firmware even better.**

bmorcelli, 2025-03-20:
Well, unfortunately I can't read your mind over the internet, so bring some light into this matter.

* What are the bugs you are facing that need to be fixed?

* What are the features you believe are needed for the firmware 

* Wdym about a more convenient menu? (I like the way it is, IMHO)

* What other modules are you talking about?

And about the rickroll thing, it is a prank inside the firmware, we won't hardcode any other script into the firmware, because it is user's reponsability, and there are tons of repos with "Flipper Zero" BadUsb payloads that work with Bruce

mamadavi, 2025-03-20:
As for the functions, as I described, it would be nice to add emulation of the Arduino code. You can also add a function for emulating the code in turn, for example, you can make it so that after executing one script, another one is automatically executed. This will help in that one script can be responsible for disabling protection, and the second for emulating malicious code

bmorcelli, 2025-03-20:
Emulation of Arduino code? You want a C++ interpreter for what? Bruce already has a JavaScript Interpreter for this purpose

> You can also add a function for emulating the code in turn, for example, you can make it so that after executing one script, another one is automatically executed.

This is solvable with a simple copy/paste at user side, no need to predict or chain-up scripts

DDLS10, 2025-03-20:
I saw this thread and about the Badusb, you think it would be possible with a NRF and a 2.4 GHz Mouse receiver ive seen it done with other Arduino platforms but wonder would it be possible  on Bruce

IncursioHack, 2025-03-21:
> As for the functions, as I described, it would be nice to add emulation of the Arduino code. You can also add a function for emulating the code in turn, for example, you can make it so that after [...]

I don't see any advantage, we have the Javascript Interpreter so you can create your own automation, besides, think about the practical effect of a BadUSB, Windows opened a PowerShell and you executed a command, if the logic of the 2nd command is to open a PowerShell and execute, you will have problems, it is more practical and easier to put all the action in a single code, but again, this is the user who needs to create it. Regarding the scripts, there are in fact a multitude of ready-made scripts for you to base yourself on and you can create your own too. Remember that Cardputer (or any other ESP32) and Bruce are tools for you to study, create proof of concept and improve your skills as a security researcher, if you have any personal need regarding functions, we suggest that you develop and send a PR to Bruce so we can evaluate whether or not to include the code.