BruceDevices/firmware #951: Adding Bruce Firmware to a Standard ESP-WROOM-32 | ESP32 DEVKIT ❗️

mamadavi, 2025-03-20:
Hello, developers, I would like to raise the topic of adding a firmware version specifically for standard esp32 boards. It would be nice to move the functionality from the screen to the web interface from the access point at 192.168.4.1 and implement part of the bruce functionality there, as in other firmware. I would also like to discuss the topic of adding a handshake interception function during network deauthentication.  I think that adding a function that will allow you to control the device via a web interface will advance the project and overtake competitors, such as cathack, which saves certain information directly to the SD card or to the device's memory. This is very important because the ESP32 has a HANDSHAKE interception function, which opens up the possibility of other attacks or testing

mamadavi, 2025-03-20:
Also handshake interception can be involved in EvilTwin attacks, which already exists in bruce. I suggest you add a handshake interception feature, as well as password authentication using the handshake itself in EvilTwin attacks, so that the victim cannot enter the wrong password, as it is implemented in professional tools.

Japanjot99, 2025-05-22:
is this feature now included? how can i access please tell
I really want to test attacks with my esp32 devkit v1, I first  checked esp32 marauder  but it doesnt have web UI

M3l0nByte, 2026-10-31:
does somebody have wroom 32 bin file?