BruceDevices/firmware #955: Bruce mobile app

AnvXprogs, 2025-03-21:
What about app for m5stick or another device with same function like flipper app

IncursioHack, 2025-03-21:
We already have an idea of ​​this genre, but it won't be soon. Creating an app that does this communication in a clean way and that works on multiple devices will take a considerable amount of time. For now, use WebUI for some functions.

pimentelleo, 2025-03-28:
Not hard to implement! We just need a frontend dev that are capable of build anything, he can use their favorite framework and we can build the application for PWA and the APK using CapacitorJS!!!

drushadrusha, 2025-04-20:
I'm up to make iOS/watchOS app if API will be developed. It's going to be fucking awesome using Bruce capabilities through the Apple Watch.

eadmaster, 2025-05-09:
Some features are already available while running the webserver via this endpoint "http://bruce.local/cm"

Docs:
https://github.com/pr3y/Bruce/wiki/Serial#how-to-send-commands

andreock, 2025-06-04:
@Pablo-Ortiz-Lopez Did you make some progress? I can help if you need. I have experience in capacitor JS + BLE to interface with ESP32 so I can help with both frontend or BLE API implementation.

pr3y, 2025-06-05:
Guys this will be a thing soon!
There is already a repo for BruceApp and it will be public, the interactions via Serial already work, the code is in Kotlin and multi-platform (desktop and android builds for now)
I was having a issue on making the flasher work via USB OTG for android to release it on fdroid and playstore! If some of the contributors wanna read the repo and tell your opnions feel free to ping me

andreock, 2025-06-06:
> the interactions via Serial already work,

Did you mean with BLESerial or USBSerial?