BruceDevices/firmware #956: RF - LoRA E220 module

Mraauuuu, 2025-03-21:
I'm using M5StickC plus 2, and trying to do some test with RF module.
I bought a lora E220 module and later read that it is not compatible (at wiki)

Is there a chance that the functionality (compability with Lora modules) will be launched in the future?

IncursioHack, 2025-03-21:
If the intention is to use the LoRa module in the RF part of Bruce, it will not be possible, despite the frequency, the form of communication is completely different.

Implementing LoRa in Bruce is a challenge, the first is to understand the practical application, where today LoRa is only used to send and receive messages, mainly in IoT environments, I have not found anything that involves hacking in this sense. Bringing all the Meshtastic firmware to Bruce will also not be possible, unfortunately.

PRSXFENG, 2025-03-22:
M5Stack LoRa module does not expose a raw radio for the firmware to use

Instead, there is a STM32 (or other) Microcontroller in between and you send specific commands to it over a serial connection

It doesn't provide enough control for Bruce or other projects like Meshtastic

It's intended purpose and the only thing it's good for is sending and receiving data or text over long distances

closed 2025-05-26 by bmorcelli