BruceDevices/firmware #959: NFC Emulatiom

biscuitaep, 2025-03-21:
Is emulating with pn523 (5321) impossible? Or is it coming soon

biscuitaep, 2025-03-21:
@bmorcelli

bmorcelli, 2025-03-21:
It is not impossible, but very limited..

I can't predict when it will come through

IncursioHack, 2025-03-21:
[image]

If you want to emulate TAGs and Cards, unfortunately it will not be possible.

IncursioHack, 2025-03-21:
If the intention is to emulate, in any way you can, my suggestion is Chameleon and similar versions, it reads, writes and emulates in 13.56Mhz and 125Khz, it has compatibility with Bruce via BLE and on mobile with several applications.

https://www.youtube.com/watch?v=ZInyYs_0YAw&pp=ygUUQ2hhbWVsZW9uIFVsdHJhIFIzY2s%3D

biscuitaep, 2025-03-21:
Thank you, im making my own pcb and i think i can use st25r3916 and write custom code for it, in chamelon ultra docs they say the nrf52 is best but its hard to use with Esp32

IncursioHack, 2025-03-21:
R3ck and I have been arguing for months over this documentation and some websites mentioning that the PN532 emulates. We considered several other chips and the like. What we discovered is that the PN532 only partially emulates, meaning it would not be possible to emulate a company badge or tags used in residential buildings and condominiums.
In addition, even though some websites say it emulates, absolutely none of them showed evidence or code for complete emulation. When we discussed it with the manufacturer, they said that the PN532 was old and they would no longer support it.

Our last research stopped at the EMVCo chips used in the Flipper ( https://www.st.com/en/nfc/st25r3916.html#overview ), as these are also used in credit card payment machines (POS).
https://docs.flipper.net/nfc
We looked for ways to purchase the chip with some pre-made schematics that could be sent to Brazil, but we had no luck. We also didn't find anything on Aliexpress/Amazon. So for now, research on this topic and testing are on hold.

If you can get an integration of st25r3916 with ESP32, please let us know in a Pull Request to integrate with Bruce, we are interested in seeing this module working.

biscuitaep, 2025-03-21:
> R3ck and I have been arguing for months over this documentation and some websites mentioning that the PN532 emulates. We considered several other chips and the like. What we discovered is that the [...]

Okay thank you so much for reply, i will try to make it but its hard because there is no emulation support on the library so its going to be a long process

closed 2025-03-26 by IncursioHack

samallx, 2025-04-12:
@biscuitaep did you make the pcb yet? i am stuck at the antenna circut because i can't find RFO2 on my chip. Also i don't know if i should include the resistors, caps named "DNP": https://docs.flipper.net/development/hardware/schematic

biscuitaep, 2025-04-12:
> @biscuitaep did you make the pcb yet? i am stuck at the antenna circut because i can't find RFO2 on my chip. Also i don't know if i should include the resistors, caps named "DNP": https://docs.flipper.net/development/hardware/schematic

Not yet, i dont have that much time.
i think you should include the pads on pcb in case of it doesnt work