BruceDevices/firmware #961: Feature request Karma Attack

Gosheto1234, 2025-03-21:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Can you add Karma attack in the future ? 

**Describe alternatives you've considered**

**Additional context**
​A Karma attack is a cybersecurity exploit targeting the automatic Wi-Fi connection behaviors of client devices like smartphones, laptops, and tablets. These devices often have a feature that remembers previously connected Wi-Fi networks and attempts to reconnect to them automatically for user convenience. This process involves the device broadcasting 'probe requests' to detect known networks in the vicinity. ​

In a Karma attack, an attacker leverages this behavior by monitoring these probe requests in public areas. The attacker then sets up a rogue Wi-Fi access point mimicking the SSID (network name) from the device's probe request. Believing this to be a trusted network, the device automatically connects to the malicious access point without user intervention. ​

Once connected, the attacker can execute various malicious activities, including:​

Data Interception: Monitoring and capturing sensitive information such as passwords, credit card numbers, and personal communications transmitted over the network.​

Malware Injection: Distributing malicious software to the connected device, potentially leading to further compromise.​

Phishing: Redirecting the user to fraudulent websites designed to steal credentials or other personal information.​

The effectiveness of a Karma attack stems from exploiting the default behaviors of wireless devices, which prioritize convenience over security by automatically seeking and connecting to known networks

mamadavi, 2025-03-23:
They don't respond to really good offers.

akira-a23, 2025-03-23:
may i add that karma attack would be incredibly hard to add. If you are really that into counter attacks get marauder it can find other esp32s and flippers