BruceDevices/firmware #967: Lilygo T-Display S3

supermaaar, 2025-03-23:
Planning to create PCB for Lilygo T-Display S3 
May I ask the GPIO for 
Up
Down
Select

Also if 
I will add the SD card reader which pin is it?

Thank you

bmorcelli, 2025-03-24:
I set up the navigation to not use extra buttons, so you can navigate, select and esc using only the 2 buttons, and it giver you 1gpio touse for other things

Sdcard isn't defined yet.. if you're planning to make a custom pcb it is a good time to determine whatever pins it should use for SD (4 pins), CC1101(+2), Nrf24(+2), IR (tx and Rx) and the I2C/UART(+2)

Bagnoni, 2025-03-30:
Could be a good idea to use the original lilygo shield configuration (1bit mode) with only 3 pins and without CS pin?

[image]

OriginalNode, 2025-03-31:
hi @bmorcelli I'm not able to select options, both buttons switch to the right or to the left

bmorcelli, 2025-03-31:
Install Beta

In the 1.9.1 you need to add a physical button on gpio 16.. nothing trivial

On Beta it uses short press to go back and forth, long press to select and esc

closed 2025-05-26 by bmorcelli