BruceDevices/firmware #977: Brucegotchi NFW on CYD

MikeOxlong83, 2025-03-24:
**Describe the bug:**
Brucegotchi loads, but all it does is sit there and say things like:
"SO ANGERY, BUT CAN'T PWN" 
"Really? This hardware? Can't PWN"
"It works on my end"
"Preparing deauth sniper"/"lasers activated deauthing"/"lasers activated! pew pew pew"
"3.14157..." (Pi)
and the lower right corner says "NOT AI", but on my flipper doing a marauder sniff for pwnagotchis detects brucegotchi

**Steps To Reproduce:**
Run Brucegotchi on CYD
- Device used: CYD ESP32-2432S028

**Expected behavior:**
Normal pwnagotchi stuff

**Actual behavior:**
The above messages loop over and over

**Additional context / logs:**

bmorcelli, 2025-03-25:
Well, you can't see with your bare eyes what it does behind the screen..

Turn you deauth packets detector on Marauder and let it sniff.. you will see brucegochi deauthing the sniffed networks while in the "pew pew pew" screens

Try to connect and disconnect from your 2.4ghz network, and you might see the handshakes counter increasing

Handshakes are saved in the SDCard in a specific folder (sorry, but the old man doesn't recall the folder name)

The "NOT AI" means that it doesn't have pwnagochi feature to try to break the handshakes, because its processing power is way weaker than rpi 0w to such a hard task..

Sometimes the screen bugs, because the sharky is working so hard on the background that it forgets how to mess with the display..

Don't expect it to be a real pwnagochi, and that's why we changed the name to Brucegochi (Community chose it, btw)

DionisioVN, 2025-03-29:
Por favor onde faço a quebra das senhas de handshakes ?

closed 2025-05-26 by bmorcelli