BruceDevices/firmware #987: UI themes sucks after a deep experience !

matoopsa, 2025-03-25:
After a deep experience of the beta version, especially UI themes, it has developed from the previous problems, but there are still clear problems that are summarized in the following points: 1. When you add an image, the background remains, for example, if you replaced the wifi icon image with another image via bruce themes build, the icon will remain there in a really annoying way. 2. You can barely apply only one image because the change will not be applied to the rest of the icons, especially if they are consecutive. 3. Suppose you put a variable fm icon for me and named it image one, then this image linked to fm will appear on the clock out of nowhere. I expected that the problem was that I chose a display resolution of 0.8. It is possible that I should lower it in order to reduce the computational load 

On this time the image was supposed to appear in the clock section but it appeared with the icon in the others option also when you flip between the icons you find that part of the image appears incorrectly I also added another image in the files section but to no avail as the image did not appear at all

[image]
[image]

tommyfart, 2025-03-25:
how do you do custom themes

IncursioHack, 2025-03-26:
Remember that themes are just a "look" for Bruce, not the crucial/essential part. It is in BETA version for users to carry out the proper tests.

To understand what is happening, I need you to inform some things and perform some actions, the whole step-by-step is described in Bruce's discord at #bug-report-rules.

The title and description look more like an emotional outburst than a bug report. Understand that Bruce is free and you can develop and correct what you think is ideal. Bugs can and will occur and we count on the tests and patience of the community, considering that we had several themes that worked correctly. Once you have attached the necessary information according to Bruce's #bug-report-rules, we will be able to give you attention to the cited case.

bmorcelli, 2025-03-26:
> 1. When you add an image, the background remains, for example, if you replaced the wifi icon image with another image via bruce themes build, the icon will remain there in a really annoying way. 
 
It is being fixed, i will submit the fix in a few minutes

> 2. You can barely apply only one image because the change will not be applied to the rest of the icons, especially if they are consecutive. 
 
idk if you used the theme builder html.. if so, you can drag all images to the pool and then you need to move the images to the right spots... so you need to move the wifi image to the wifi spot, ble image to ble spot and so on.. if you have a better idea, change the code of the theme_builder.html and open a PR

>  3. Suppose you put a variable fm icon for me and named it image one, then this image linked to fm will appear on the clock out of nowhere. I expected that the problem was that I chose a display resolution of 0.8. It is possible that I should lower it in order to reduce the computational load

the 0.8 is not a resolution, but compression rate of the image.. it will is meant to reduce computational load without reducing the width/height of the image.

closed 2025-03-27 by pr3y

matoopsa, 2025-03-28:
Regarding the problem report I have informed the officials about the problem and it is not a major crisis because it is not related to Bruce’s goal but it must be mentioned in the beta stage so that we can solve the problems nothing more  and I apologize if any of you thought it was an emotional speech I may have expressed it incorrectly but our goal remains to develop open source systems and help with them.

> Remember that themes are just a "look" for Bruce, not the crucial/essential part. It is in BETA version for users to carry out the proper tests.
>
> To understand what is happening, I need you to [...]

Regarding the problem report I have informed the officials about the problem and it is not a major crisis because it is not related to Bruce’s goal but it must be mentioned in the beta stage so that we can solve the problems nothing more  and I apologize if any of you thought it was an emotional speech I may have expressed it incorrectly but our goal remains to develop open source systems and help with them.

bmorcelli, 2025-03-28:
Hey man.. I sent a fix for that, and I need your feedback about it..

It is now clearing the screen each time it changes the image to avoid this "ghosting" image from the previous menu..

So I appreciate if you could test it

matoopsa, 2025-03-30:
> Hey man.. I sent a fix for that, and I need your feedback about it..
> 
> It is now clearing the screen each time it changes the image to avoid this "ghosting" image from the previous menu..
> 
> So I appreciate if you could test it

Ok bro i tried it and it worked perfectly but I want to point out that the image must be in PNG encoding as no other encoding will work Also, the ripple problem has been solved, but I will show you a simple error that can be worked without

[image]

 but it must be pointed out