BruceDevices/firmware #991: Chamelon ultra dont work

Jojorel, 2025-03-27:
**Describe the bug:**
Dont show anny tag or similar, conect but dotn show 

**Steps To Reproduce:**
Conect but dont work
- Device used: 

**Expected behavior:**

**Actual behavior:**
Dont show the behavior

**Additional context / logs:**

IncursioHack, 2025-03-27:
Could you provide some details or a video? I just tested my Chameleon here and it connected without any problems.

mileshkin, 2025-03-27:
Turn off BLE Auth in Chamellion - it will solve issue.

Jojorel, 2025-03-27:
It is conected, when im try to read something and just keep blink a white light and dont scan.

Jojorel, 2025-03-27:
> Turn off BLE Auth in Chamellion - it will solve issue.

How im get BLE AUTH in chamellion ?

IncursioHack, 2025-03-27:
Have you ever connected Chameleon to your smartphone? If so, could you remove it from the list and turn off Bluetooth on your smartphone? In some cases, Chameleon may remain connected to the smartphone and not connect to the Cardputer.

Can you send us a video of how you are trying to connect to Chameleon?

Jojorel, 2025-03-27:
https://github.com/user-attachments/assets/b6556edf-81ee-4645-bfaa-71c7fbd96876

This way, im not conect ever my phone to chameleon

IncursioHack, 2025-03-27:
> WhatsApp.Video.2025-03-27.at.20.42.08.mp4 
> This way, im not conect ever my phone to chameleon

If possible, please post a video of the moment you connected to the Chamaleon (opening the MENU on Bruce, turning on the Chameleon, etc.). Out of curiosity, is the tag you entered 125Khz or 13.56Mhz? If it is 13.56Mhz, the correct selection is HF Scan, LF Scan is only for 125Khz.

If you have a computer nearby, before doing the entire connection procedure with the Chamaleon, connect it and collect the SerialLog, this helps a lot in the analysis.

Jojorel, 2025-03-28:
Im conect all the options im have, on this video, how im put the serial log to view that ?

IncursioHack, 2025-03-28:
Collecting Logs (Serial Log)

How to get the serial log:
• Connect your ESP32 device to the computer via USB cable.
Install the required software:
• If you haven’t done so yet, download and install the Arduino IDE or PlatformIO.
Configure the serial monitor:
• In Arduino IDE: Open the Arduino IDE, connect the ESP32, and select the correct port under Tools > Port.
• Go to Tools > Serial Monitor or press Ctrl + Shift + M to open the serial monitor.
• Make sure the baud rate is set to 115200 (or the same as you used when programming the ESP32).
Collecting the log:
• Now, after reproducing the error, check what appears on the serial monitor. Copy the logs and paste them into a text file.
• If the device crashes or restarts, look for error messages or crashes in the log.

Another option:
* Go to https://web.esphome.io/
* Connect your device
* Select Serial

Jojorel, 2025-03-28:
https://drive.google.com/file/d/1F7M7iukPsKWrkCIREJgsUemE3wpJiLGk/view?usp=sharing

IncursioHack, 2025-03-28:
> https://drive.google.com/file/d/1F7M7iukPsKWrkCIREJgsUemE3wpJiLGk/view?usp=sharing

Now I can understand it much better, the battery % is displayed, which means it was actually connected, the FACTORY RESET command also worked apparently, it just couldn't read the TAG.

There is one of the devs who has exactly this Chameleon model, I'll see if he can do a test.

While I'm at it, do you have any other tags?

Jojorel, 2025-03-28:
Yes im tested with all im have and cards to :( nothing ready

Jojorel, 2025-03-28:
Thanks alot for your help

IncursioHack, 2025-03-28:
I just noticed you are on version 1.9.1, could you test it in BETA?

1 - Please install the BETA version, as shown in the image at:
https://bruce.computer/flasher

2 - If you want the Binary to insert into your SD and run with the Launcher, you can download it here:
https://github.com/pr3y/Bruce/tree/WebPage/BetaRelease
OR
https://github.com/pr3y/Bruce/blob/WebPage/BetaRelease/Bruce-m5stack-cardputer.bin

3 - Finally, if it doesn't work, delete the LittleFS bruce.conf file and SD Card
• Backup: Before deleting anything, make sure to back up any important data. You can back up the files directly from the WebUI.
Delete bruce.conf files:
• Access the WebUI of the Bruce from another device (computer, laptop, or smartphone).
• Go to the LittleFS menu.
• Locate and delete the bruce.conf files from LittleFS (internal storage).
• Go to the SDCard menu (if present).
• Locate and delete the bruce.conf files from the SD Card (external storage), if present.
• After deletion: Restart the Bruce and test the device again to check if the problem persists.

Jojorel, 2025-03-28:
im do all the things you ask, but not work, im reinstal on beta mode, nothing, delete on sd, webui to delete the bruce.conf in all ways. but not work :0

closed 2025-05-26 by bmorcelli