BruceDevices/firmware #993: Liligo T-Display S3 SD card

seventaiger, 2025-03-27:
SD card works with Bmorcelli's launcher wired as SD_MMC as previously coded but did not work on Bruce FW. I now see on the new revision for Bruce that it is to be wired as SPI. If I change SD connection to SPI, Launcher will no longer work? Any way to make both work or is the Liligo S3 unable to run the SD card and other devices using SD_MMC?

Grzybekrk, 2025-04-04:
I have the same :(

Huzzla101, 2025-04-04:
Just reflash firmware without sd card inside and it should be ok BUT LIsTEN

RF SCANNER IS STARTING TO BE MAGICAL no doubt... , in just phew months its
already in top spot.. Bruce suddenly rules on over 10 different devices...
BUT what we really need now that the NRF24 is FIIIRRREEEEEE  with new
jammerl modules etc.. Is really a authenticator that can sniff all the raw
traffic , we already have the spectrum that kind of do the same just it
doesnt write the raw format.. As all NRF24 have a main device that wsend
out connection signal and other and a remote etc that do the same back.. So
Last thing needed now is a good NRF24 sniffer/autheenticator for 0,5mb and
1mb and also 2mb traffic with CRC and ESB, possibility to read in encrypted
mode to

WHY IT IS VEERY IMPORTANT.. ? Well only one other device have gone so far
and that our little brother Flipper.. And they where WAY ahead of us just
months ago but now with Mr Pirata and all his God Coders Bruce is starting
to become the talking deal for almost every device out there .... AND the
sniffer is the reason mouse jacker was made and possible becouse we could
read the signal from the usb mouse dongle  sendt to the mouse and also
other way there of we could in the end send the mouse usb dongle any
commands as a BAD USB.... And also its possible to sniff ANY DEVICE that
operates on 2,4ghz with a nr4 antenna and sniffer..
AND since you have the best jammer we must have also a authenticator....
Worst case i can ask Vadim , the magician who made nrf24sniffer and mouse
jacker etc for flipper. he is mastermind of this... And maybe he want help
us programm a little...

bmorcelli, 2025-04-10:
You can test it here: https://bmorcelli.github.io/Bruce/flasher

Idk if you are using the Willy firmware, but pins were set up for it (I have one to test myself)

[image]

seventaiger, 2025-04-11:
Using new firmware linked above did not resolve SD card issue. Bruce still does not seem to see the SD card.

bmorcelli, 2025-04-11:
One thing that I noticed, for this SD_MMC, os that it must start with the SD plugged on..

If you plug the SD with bruce running, it won't mount, so you meed to restart..

Other important thing is that the SD Card must be formatted in FAT32 (preference for SDHC lesser than 32Gb)

seventaiger, 2025-04-12:
Tried everything you suggested with no success. Is SD card wiring pinout same as Launcher?

bmorcelli, 2025-04-12:
The flasher I pointed is the one from my brench, this solution wasnt merged yet...

Try to download this binary and install with launcher:

https://github.com/bmorcelli/Bruce/blob/WebPage/BetaRelease/Bruce-lilygo-t-display-s3-touch-mmc.bin

seventaiger, 2025-04-12:
Thank you very much. That solved the issue. Bruce now runs from Launcher's SD and also recognizes the SD card.

closed 2025-04-12 by seventaiger