BruceDevices/firmware #995: sd card issue

johnalpha911, 2025-03-27:
**Describe the bug:**
restarts from sd card for no reason

**Steps To Reproduce:**
when i put the sd cadr in and i try to access it it justs starts booting again for no reason
- Device used: 
- cardputer

IncursioHack, 2025-03-27:
Please give more details, inform:

Which version of Bruce are you using?

Have you tested it in the most current BETA version?

Which model of SD card are you using?

Did it work in previous version?

Is the SD Card properly formatted (FAT32) and is it smaller than 32GB?

Could you provide the LogSerial at the time of the problem?

johnalpha911, 2025-03-27:
Im using 1.9.1, i haven't tested in other versions. I use m5 launcher. It is 16gb and it is properly formated

IncursioHack, 2025-03-27:
**1 - Please install the BETA version, as shown in the image at:**
https://bruce.computer/flasher

[image]

**2 - If you want the Binary to insert into your SD and run with the Launcher, you can download it here:**
https://github.com/pr3y/Bruce/tree/WebPage/BetaRelease
OR
https://github.com/pr3y/Bruce/blob/WebPage/BetaRelease/Bruce-m5stack-cardputer.bin

**3 - Finally, if it doesn't work, delete the LittleFS bruce.conf file and SD Card**
• Backup: Before deleting anything, make sure to back up any important data. You can back up the files directly from the WebUI.
Delete bruce.conf files:
• Access the WebUI of the Bruce from another device (computer, laptop, or smartphone).
• Go to the LittleFS menu.
• Locate and delete the bruce.conf files from LittleFS (internal storage).
• Go to the SDCard menu (if present).
• Locate and delete the bruce.conf files from the SD Card (external storage), if present.
• After deletion: Restart the Bruce and test the device again to check if the problem persists.

**4 - If none of the cases work, we need more details of the problem, such as:**

A - Has your SD card worked with another version of Bruce before?

B - We need LogSerial, if you don't know how to get it, we have all the details of a BugReport here:
https://discord.com/channels/1264994142769909854/1354121493478641805

johnalpha911, 2025-03-27:
now it says fail mounting sd card even if i reset it

johnalpha911, 2025-03-27:
and it still does the same

bmorcelli, 2025-03-27:
Check if you don't have too many files on the root folder of your Sdcard..

Maybe it is overflowing the file list or exceeding the memory limit

But as Incursio wrote.. It is working here on my side too, and to help you we need more details

johnalpha911, 2025-03-27:
the sd card is thishttps://www.aliexpress.com/item/1005001617961938.html?spm=a2g0o.order_list.order_list_main.5.21ef18023gIIeZ
its formated as fat32it works in the launcher but i dont know how to check log serial

johnalpha911, 2025-03-27:
and it doesnt work with less files either

bmorcelli, 2025-03-27:
I limited the file list to 250, same as Launcher and before..

Can you test it please?

johnalpha911, 2025-03-28:
250 files in the root folders or 250 at any place in the sd

johnalpha911, 2025-03-28:
And it doesn't work even with 35 files

johnalpha911, 2025-03-28:
I use m5 launcher though and one question which files are needed for the sd because the only file created by a firmware is config.conf and i can access the files but when i enter a folder it crashes

Huzzla101, 2025-03-28:
Since you have a 32gb you have to put the whole bruced firmware on it...
Thats rule nr 1 .. :p fill it up make no changes to it. It will consist of
a root with folders and some programming files at end, and when you make a
RF or WIFI or even IR file then Bruce will make sure of that you will get a
special made map appear in root with ur stored things..

AND it even have a sd-cardfiles map that you can BLAST in all your big well
dressed DATABASES of all the IR codes out there and subGHZ/RF , and YES
almost ALL the used big DB that flipper USES , YOU CAN simply download it
and put it into the sd-folder then you have IT ALL !!!!! I must use a week
or 3 to just go thru all of my shiet that i have on Bruce... He just laying
in the water waiting...  And your attached SD card reader uses 6 pinouts?
you made sure its intact? have it ever read any SD cards before ??

Huzzla101, 2025-03-28:
Ahh you use the Launcher... haha  ok my bad. well anyone else having SD
card problems please read above !! =))))

johnalpha911, 2025-03-28:
So the launcher is causing the problem because it didn't work without the launcher either

Huzzla101, 2025-03-28:
It didnt work without launcher to ? You mean the SD card ? Have you got
that to work before the card ? Try reflash launcher again without the sd
card inside , from website... And enter Sd card AFTER you have successfully
RE installed FW

johnalpha911, 2025-03-28:
wait sorry i meant m5launcher not bruce launcher

Huzzla101, 2025-03-28:
Yes re install it without the SD card inside... You using Cardputer ? Then
enter SD card after you have re installed m5launcher  and check if it can
mount it.. Have you ever mounted another card successfully before THO ?

johnalpha911, 2025-03-28:
It works in m5 launcher but it doesn't work in Bruce. Why?

Huzzla101, 2025-03-28:
Have you tried to change to a different SD card tho? Reflash ur Bruce and
put the needed SD card files into a new card and try again as this is just
a luxury problem that involves you , thereof it must be some small details
at your stash u need solve,  as everyone else , like my self even got a
64GB sdcard to work with bruce. And i even used a NONE BRAND 32gb card and
all works ...BTW have you ever got a SD card to work with bruce yet ???

johnalpha911, 2025-03-28:
where can i find the needed sd card files?

johnalpha911, 2025-03-28:
now i re formated it and it started bootlooping

johnalpha911, 2025-03-28:
update it works with another sd card im so happy right now

Huzzla101, 2025-03-28:
BRUCE team never plays around my friend.... When you happy We Happy!! =)

closed 2025-03-28 by bmorcelli