BruceDevices/firmware #997: Bruce Bluetooth

herpaderr, 2025-03-27:
The device don’t appear on my phone when I’m trying to use bad ble or the media thing. This issue appeared on the latest beta and the one before the latest beta. It appears on my phone on Bruce version 1.9.1

herpaderr, 2025-03-27:
Still ain’t working, the device I’m using is the m5stickc plus 2

IncursioHack, 2025-03-27:
I just tested it on both StickC Plus 1.1 and Cardputer, in both I was able to identify, connect and use (as shown in the images below).

Im currently using the latest BETA version ( 93c3502 ) :
https://bruce.computer/flasher

Try deleting bluetooth from your smartphone.

[image]

[image]

[image]

[image]

[image]

[image]

EsdeathOG, 2025-03-27:
Still ain't working for me, but it appeared on my android tablet tho but not on my apple devices

[image]

IncursioHack, 2025-03-27:
In fact my cell phone is android and it appears to me, just as it appears on your Android, I don't have an IOS to test. At this point I'll owe you that.

bmorcelli, 2025-03-27:
There was a small change from 1.9.1, that was about setting a different "UUID" for each device, allowing you to have more than one Bruce connected to your device.

Try to "forget" the older BLE device in your phone, and then scan it again... Tell me if it works

herpaderr, 2025-03-27:
Still ain’t working

[image]

EsdeathOG, 2025-03-27:
Did this get fixed in the newest beta?

bmorcelli, 2025-03-27:
Try it out and tell us

herpaderr, 2025-03-27:
Nope still not working

bmorcelli, 2025-03-28:
Can you test the last build on BetaRelease?

EsdeathOG, 2025-03-28:
Still ain't working

EsdeathOG, 2025-03-28:
It still only works on 1.9.1 for me

ben7147, 2025-04-23:
Same problem, latest build ):

HenryTheAddict, 2025-04-23:
Brucetooth.

ben7147, 2025-04-23:
Does someone know where is the keyboard layout file in src code?

closed 2025-05-26 by bmorcelli