BruceDevices/firmware #1002: 125 khz rfid

DDLS10, 2025-03-28:
**Is your feature request related to a problem? Please describe.**

Bruce does not support 125khz RFID

**Describe the solution you'd like**

Adding a feature with the RFI coils connected to gnd and a gpio pin

**Describe alternatives you've considered**

Using a coil for 125 khz

**Additional context**

here is a example of an esp32 with 125 rfid https://github.com/Crypter/ESP-RFID/blob/master/README.md

DDLS10, 2025-03-28:
RFID coils *

rennancockles, 2025-03-28:
Bruce does support reading rfid 125kHz, but only using an RDM6300 module

DDLS10, 2025-03-28:
Yes I know but I was talking about emulating rfid

IncursioHack, 2025-03-28:
You need to specify what you need, your initial text was about not supporting, now it's about emulating. It is possible to emulate 125khz (and 13.56Mhz) with Bruce through Chameleon, in this case, this integration already exists and is functional.