BruceDevices/firmware #1007: Station deauth crashing

Piecyk228, 2025-03-29:
**Describe the bug:**
Station deauth reboots device

**Steps To Reproduce:**
Go to wifi menu, choose scan hosts, selcet your wifi, select ip, select staion deauth option
- Device used: Lilygo t embed c1101, m5stick c plus 2

**Expected behavior:**
Station deauth happening

**Actual behavior:**
device reboots

**Additional context / logs:**
firmware version beta cc1f9fc and normal 1.9.1 release

IncursioHack, 2025-03-29:
Hello, first of all I would like to thank you for testing and informing both the devices and the version, this helps a lot, finally if you could perform the test again and collect the SerialLog it would help a lot. :)

Piecyk228, 2025-03-29:
Of course. I'll do so when I have time and comment results

Piecyk228, 2025-03-29:
I'll be doing so on lilygo t embed cc1101 release 1.9.1 and m5stick c plus 2 release 1.9.1

Piecyk228, 2025-03-29:
i didnt copy my ip or passwords here they are replaced with redacted, at least i tried to do so if there is sensitive data here inform me please: [ 20424][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 255
Selected: Connect Wifi
[ 28237][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 28279][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started
[ 28285][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[ 35306][V][WiFiGeneric.cpp:384] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 17
[ 35315][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 1 - SCAN_DONE
Selected: Bielik
[ 38998][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: REDACTED, MASK: REDACTED, GW: REDACTED
[ 39461][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Bielik, BSSID: da:b3:70:1c:bd:91, Channel: 1, Auth: WPA2_PSK
[ 39474][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 39487][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got New IP:REDACTED
[ 39495][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 39503][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: REDACTED, MASK: REDACTED, GW: REDACTED
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 3,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 10,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "REDACTED"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "REDACTED"
  },
  "evilWifiNames": [],
  "bleName": "KeyboardT",
  "wifi": {
    "Bielik": "REDACTED"
  },
  "irTx": 2,
  "irTxRepeats": 0,
  "irRx": 1,
  "rfTx": 8,
  "rfRx": 18,
  "rfModule": 1,
  "rfFreq": 433.92,
  "rfFxdFreq": 2,
  "rfScanRange": 3,
  "rfidModule": 1,
  "mifareKeys": [],
  "gpsBaudrate": 9600,
  "startupApp": "",
  "wigleBasicToken": "",
  "devMode": 0,
  "colorInverted": 1,
  "disabledMenus": [],
  "qrCodes": [
    {
      "menuName": "Bruce AP",
      "content": "WIFI:T:WPA;S:BruceNet;P:REDACTED;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
[... 153 lines cut ...]
    "io0": 3,
    "io2": 38
  },
  "NRF24_Pins": {
    "sck": 11,
    "miso": 10,
    "mosi": 9,
    "cs": 44,
    "io0": 43
  },
  "SDCard_Pins": {
    "sck": 11,
    "miso": 10,
    "mosi": 9,
    "cs": 13,
    "io0": -1
  }
}[  1484][I][config.cpp:266] saveFile(): config file written successfully
[  1518][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
E (1573) rmt: rmt_set_gpio(530): RMT CHANNEL ERR
E (1574) rmt: rmt_config(686): set gpio for RMT driver failed
E (1574) rmt: rmt_set_gpio(530): RMT CHANNEL ERR
E (1577) rmt: rmt_config(686): set gpio for RMT driver failed
[  1598][E][vfs_api.cpp:105] open(): /littlefs/boot.jpg does not exist, no permits for creation
[  1615][E][vfs_api.cpp:105] open(): /littlefs/boot.gif does not exist, no permits for creation

Piecyk228, 2025-03-29:
also I noticed that when i start webui on t embed cc1101 using release 1.9.1 it restarts too but only when i go on the website in borwser maybe i wont provide log because it has a lot of sensitive data

Piecyk228, 2025-03-29:
This log comes from the t embed cc1101

bmorcelli, 2025-03-29:
> also I noticed that when i start webui on t embed cc1101 using release 1.9.1 it restarts too

This WebUI problem was already fixed..

the station deauth issue is something different.. need to check why it is happening.. thank you for the log!

Piecyk228, 2025-03-29:
No problem. Have a good day or night or whatever since we are probably in different time zones :)