BruceDevices/firmware #1008: M5StickC Plus2 Shared SPI CC1101 & SD Card Issues

CodeHedge, 2025-03-29:
**Describe the bug:**
1. When using the CC1101 in conjunction with the SD card, it works until a read/write operation. The CC1101 becomes unresponsive and SD Card continues to work. The CC1101 with work again as soon as you eject the SD card. 

2. is when loading a Sub file from the SD card to transmit, it hangs until ejecting the SD, and then the signal is transmitted.

3. I can read a signal, and save successfully, but it will hang on the save success message. I force restart the device and the file saved fine.
 
**Steps To Reproduce:**
Connect both CC1101, Transistor, and SD card according to schematic. Boot the device and configure the Tx/Rx/Module settings. Verify CC1101 functionality using spectrum. Back out of RF menu and open files -> SD Card. You should be able to read files just fine. Back out then go back to RF and try to open spectrum. You will get a blank screen until ejecting the Sim. The Spectrum will immediately appear.

This behavior can be replicated system wide by performing any read/write operation on the SD card. The CC1101 will stop working until SD is ejected. This bug will surface anytime this condition is met.

- Device used: 
M5StickC Plus2
Custom board wired to schematic.

**Expected behavior:**
CC1101 should function after a read/write operation has finished.

**Actual behavior:**
SD card needs to be ejected before CC1101 can operate again.

**Additional context / logs:**
No

CodeHedge, 2025-03-29:
I suspect the SD card is somehow still communicating preventing communication to the CC1101. Since the device cannot send a command to the CC1101, the signal will never trigger the transistor that cuts of communication to the SD card.

Maybe there is a way to unmount or tell the SD card to stop communicating so the CC1101 can operate again.

That's my tinfoil hat theory

IncursioHack, 2025-03-29:
Hello, I really appreciate the details, if you could also include the Bruce version you are using, update to the latest BETA version ( https://bruce.computer/flasher ) and SerialLog, it would help a lot.

Just out of curiosity, when you went to RF > CONFIG > MODULE > CC1101 (Shared SPI), is this option visible on the screen and in bruce.conf?

CodeHedge, 2025-04-04:
> 
> 
> 
> > I suspect the SD card is somehow still communicating preventing communication to the CC1101. Since the device cannot send a command to the CC1101, the signal will never trigger the transistor that cuts of communication to the SD card.
> > 
> > Maybe there is a way to unmount or tell the SD card to stop communicating so the CC1101 can operate again.
> > 
> > That's my tinfoil hat theory
> 
> Hello. I had the same problem and I have solved it through a micro switch. I put the link of my steps. All the best
> https://github.com/pr3y/Bruce/issues/249#issuecomment-2757291123

I suppose that's an option but i thought the whole reason to even implement the transistor was to avoid the hacky workarounds. Hopefully the dev can revisit this because as of now CatHacks implementation (although less elegant) is a better experience. You can read and then transmit with no intervention on that firmware. Write is different for some reason but still less manual tinkering overall.

Alberto0581, 2025-04-04:
Hi, this is the easiest way I've ever had without inserting or removing the microSD card. I tried different resistor values ​​and swapped for a different transistor. I haven't seen any problems with adding a mini switch. I'm just starting out with this vulnerability issue, and the developer and collaborators are true geniuses. And on top of that, it's all free to download. What more could I ask for? I'd just like to repeat my thanks and my cooperation in any way I can. Best regards.

CodeHedge, 2025-04-04:
I get it, but Cathack is easier and better implemented at this current moment. This firmware is better in every other regard. We all appreciate the devs. Your work around is great too! But, there is still an issue with bruce's current implementation of the shared SPI between the SD and CC1101 on the M5stick. **I do not believe the micro switch is a good long term solution** and after speaking with the dev on discord - I get the feeling they feel the same way. They wouldn't have attempted the transistor method if they were ok with workarounds. Cathack does something similar with rebooting the device at the end of a write operation. No UI hangups/reboots either on SD reads and switching back to CC1101.

If the switch is an acceptable solution, then the UI hangups should be updated at the very least to time out or pop something up when using shared SPI to use the button. The wiring diagram would need to be updated too.

At the current time, this feature is broken without a workaround.

Alberto0581, 2025-04-05:
Hello, thanks for your reply. On Monday, if I can, I'll start looking into the electronics (because I'm very green with programming) to see if with the current scheme, by modifying some value of some component without adding any more components, like the switch in my case, we can make the current firmware work. I'll focus on the microSD cards, in case depending on the brand or model (for reasons I don't know or I'm talking about out loud) it could cause the transistor to not perform the reset function properly.

bmorcelli, 2025-05-07:
Updated the Schematics.. the SDCard breakout doesn't have the CS pin pulled up, so I added a 10k resistor connecting CS to 3,3Vcc, so it fails to work.

[image]

Alberto0581, 2025-05-07:
Perfect, thank you very much. Modified and working.

closed 2025-05-26 by bmorcelli