BruceDevices/firmware #1010: esp 2432S028 when connect t owebui reboot auto

samexit23, 2025-03-29:
**Describe the bug:**

**Steps To Reproduce:**

- Device used: 

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

IncursioHack, 2025-03-29:
Hi, just a tip: when you open an issue, please provide us with some more details. We have several tips on how to open an issue here on Discord:
https://discord.com/channels/1264994142769909854/1354121493478641805

Just to let you know:
1 - What version of Bruce are you using? 
2 - Do you have any other modules connected? 
3 - Do you have an SD card? 
4 - Can you provide a SerialLog so we can analyze the case?

Also, we are aware of a bug in WebUI that was fixed in the BETA version, could you test it in the BETA version?

1 - Please install the BETA version, as shown in the image at:
https://bruce.computer/flasher

2 - If you want the Binary to insert into your SD and run with the Launcher, you can download it here:
https://github.com/pr3y/Bruce/tree/WebPage/BetaRelease

Without the necessary information and tests, it is difficult to analyze the case.

bmorcelli, 2025-03-30:
It has been fixed and is available in the Beta