BruceDevices/firmware #1015: T-Embed CC1101 Boot audio not working

th1s-david, 2025-03-30:
**Describe the bug:**
Inserting boot.mp3 into the device doesn't cause any audio to play on boot

**Steps To Reproduce:**
Attempt to put boot.mp3 into root directory of LittleFS then restart
- Device used: Lilygo T-Embed CC1101

**Expected behavior:**
Provided audio would play out

**Actual behavior:**
Silence and depression 

**Additional context / logs:**
Bruce version: a4aec1f (Beta)

Playing the audio through LittleFS works normally

th1s-david, 2025-03-30:
After reviewing the source it turns out that it checks for .wav instead of .mp3, but it plays the audio *after" the boot animation, would be nice if it could play during it

th1s-david, 2025-03-31:
After some more digging:

(all in /src/main.cpp)
Lines 279 to 293

```
void startup_sound() {
#if !defined(LITE_VERSION)
#if defined(BUZZ_PIN)
  // Bip M5 just because it can. Does not bip if splashscreen is bypassed
  _tone(5000, 50);
  delay(200);
  _tone(5000, 50);
  /*  2fix: menu infinite loop */
#elif defined(HAS_NS4168_SPKR)
  // play a boot sound
  if(SD.exists("/boot.wav")) playAudioFile(&SD, "/boot.wav");
  else if(LittleFS.exists("/boot.wav")) playAudioFile(&LittleFS, "/boot.wav"); //we should add a check for boot.mp3 too as stated in faq OR change faq to say "boot.wav" instead?
#endif
#endif
}
```

Lines 357 to 360

```
if (!bruceConfig.instantBoot) {
    boot_screen_anim();
    startup_sound(); //i dont know c++ but perhaps its possible to run these 2 functions at once? or run audio first without waiting for it to finish then starting boot_screen_anim function? 
  }
```

th1s-david, 2025-04-02:
don't you complete my unfixed issue. im joking of course, but I'd love for the faq to at least be fixed to say "boot.wav" instead of "boot.mp3"
(https://github.com/pr3y/Bruce/wiki/FAQ#how-do-i-customize-bruce)

bmorcelli, 2025-04-02:
Done