BruceDevices/firmware #1020: M5stick as Esp32 marauder for flipper zero

AnvXprogs, 2025-04-01:
Many people after buying M5stick buy flipper zero but official Esp32 marauder is expensive, can they use their old stick as Esp32 marauder

IncursioHack, 2025-04-01:
Hello, good morning, I was a little confused. The M5Stack StickC has a Marauder firmware (as per the image).

[image]

Could you bring more details about what you are thinking?

AnvXprogs, 2025-04-01:
i mean there is not wifi in flipper zero, but can flipper zero use m5stivk for wifi module

bmorcelli, 2025-04-01:
> i mean there is not wifi in flipper zero, but can flipper zero use m5stivk for wifi module

Possible it is, but you need to map the UART to the grove pins in the Marauder project, build it locally, and connect them into the flipper gpio..

I've seen a friend of mine doing it, but have no details..

bmorcelli, 2025-04-01:
> I've seen a friend of mine doing it, but have no details..

closed 2025-04-03 by bmorcelli