BruceDevices/firmware #1029: Cant add multiple signals in ir read > custom read

XimKa123, 2025-04-02:
Device: T-Embed CC1101 
Firmware: Bruce Beta aa513f1
Issue: If I choose Ir read > Custom read, then if i try to capture multiple  signals it will just force me to save it even if i try to cancel it to make another signal

pimentelleo, 2025-04-03:
What you mean about "multiple signals"?
Did you are saving what you read into a button pressing the next button? Can you please send a video on discord, on the #ir channel, and tag me please. 
I am considering you are not totally following the setps

XimKa123, 2025-04-03:
Like what i meant is:
When i captured first ir signal its power button and then when i try to do other ir signal for example its input button i just get forced to save .ir file with only one ir signal and its power button and also if i try to cancel this i would be still forced to create new ir file with first ir signal

PR #1036 (2025-04-07): IR Read: Fix save issue in Custom Read
Since https://github.com/pr3y/Bruce/commit/88f1dfc8c33f723f44f8b4cfb931ba4048a7dccf, the size was incorrectly calculated, I’ve fixed it here (this fixes #1029).
I also added a `quickloop` mode check to prevent saving the device after x signals when using the Custom Read mode.

Fix #1029
closed 2025-04-07 by pr3y