BruceDevices/firmware #1030: Capturing signals that dont exist

tegtheevi, 2025-04-02:
**Describe the bug:**
 when i chose a fixed frequency in Copy/scan for example 433.92mhz the cc1101 captures random signals and when i touch the antenna it keeps capturing random signals i tried putting the device down and not touching it but it happens but not like when u touch the antenna i tried changing the antenna but the problem keep happening with the long antennas only the short ones (3db) dont have the same problem 
But when i chose all ranges it doesnt happen but it doesnt capture all signals it always skips 1 (captrues 1 skips 1 captures 1 skips 1 ) 
I checked the wiring and its all good 
I changed multiple antennas and the problem keeps happening with the long antennas 

**Steps To Reproduce:**

- Device used: cyd 2 usb 
In scan/copy select fixed frequency for example 433.92mhz and touch the long antenna that comes with the cc1101 

**Expected behavior:**
 the device doesnt glitch and captrues random signals that dont exist when touching the antenna ( long antenna )

**Actual behavior:**
 the device  glitches and captrues random signals that dont exist when touching the antenna ( long antenna )

**Additional context / logs:**

Huzzla101, 2025-04-04:
> Just as a note, touching the antenna can change something:
> https://www.youtube.com/watch?v=ZrIf0PArZu4
>
> Another case is that there are fake antennas, on Bruce's Discord I have
> already posted some X-ray images of several antennas demonstrating some
> points.
>
> Your problem is extremely specific, I don't know if we will be able to
> reproduce it and if so, if it is a hardware issue or if there is something
> to be done in the software.

RF SCANNER IS STARTING TO BE MAGICAL no doubt... , in just phew months its
already in top spot.. Bruce suddenly rules on over 10 different devices...
BUT what we really need now that the NRF24 is FIIIRRREEEEEE  with new
jammerl modules etc.. Is really a authenticator that can sniff all the raw
traffic , we already have the spectrum that kind of do the same just it
doesnt write the raw format.. As all NRF24 have a main device that wsend
out connection signal and other and a remote etc that do the same back.. So
Last thing needed now is a good NRF24 sniffer/autheenticator for 0,5mb and
1mb and also 2mb traffic with CRC and ESB, possibility to read in encrypted
mode to

WHY IT IS VEERY IMPORTANT.. ? Well only one other device have gone so far
and that our little brother Flipper.. And they where WAY ahead of us just
months ago but now with Mr Pirata and all his God Coders Bruce is starting
to become the talking deal for almost every device out there .... AND the
sniffer is the reason mouse jacker was made and possible becouse we could
read the signal from the usb mouse dongle  sendt to the mouse and also
other way there of we could in the end send the mouse usb dongle any
commands as a BAD USB.... And also its possible to sniff ANY DEVICE that
operates on 2,4ghz with a nr4 antenna and sniffer..
AND since you have the best jammer we must have also a authenticator....
Worst case i can ask Vadim , the magician who made nrf24sniffer and mouse
jacker etc for flipper. he is mastermind of this... And maybe he want help
us programm a little...

GthiN89, 2025-04-07:
https://github.com/pr3y/Bruce/pull/1050/commits/c91202363e76890b90b8a5d4142610f6ec51e3c1

tegtheevi, 2025-05-26:
> same

I was able to point out the issue just yesterday if you want i can help you