BruceDevices/firmware #1031: CYD ESP32-2432S028R

mijogenbot, 2025-04-02:
Can not get Bruce to launch screen gos white ? Help please?

bmorcelli, 2025-04-02:
Picture?

Goes white how?

RoToSpLaSh, 2025-04-02:
Set the color og UI in the config menù, click INVERTED COLOR

mijogenbot, 2025-04-02:
I apologize I do not have the device with me I should have asked when I had the device I'll send some pictures when I get home

mijogenbot, 2025-04-02:
[image]

mijogenbot, 2025-04-02:
This is what I get after Flasg

mijogenbot, 2025-04-03:
[image]

PRSXFENG, 2025-04-03:
Please ensure you have flashed the correct version, from the LED being on top, this is a 2.4inch version, as such, it is NOT a CYD-2432S028

Please try
CYD-2432W328R or 2432S024R

Note: touch may not work

mijogenbot, 2025-04-03:
Correct it's not a CYD-2432S028, as I put it title to be clear it's a CYD-2432S028R, Hence the questions, I was able to flash and get colors correct touch works everything works, my question now, is there a way to adjust the size so the full screen is filled?

bmorcelli, 2025-04-03:
The R in the end is just to differentiate the touchscreen, the 2432S028 env is the R version..

What is the width and height of your board? It is not a 320x240 (standard for this kind of board) otherwise it would be showing fine

mijogenbot, 2025-04-03:
It's a 3.5" screen.

bmorcelli, 2025-04-03:
Yeah.. this unrendered margin is happening because the screen is 480x320pixels

mijogenbot, 2025-04-03:
Ok is what it is kinda thing, everything works to me that's what matters

edwardtich1, 2025-05-18:
The screen background is white and the inscriptions are not visible
I tried the regular firmware and INVERTED COLOR
Board ES32-2432S028R