BruceDevices/firmware #1041: GREAT ENGINEER & PROGRAMMER MR PIRATA DONT HAVE A T EMBED CC1101 AS HE WORKS WITH OVER 10 DIFFERENT PLATFORMS

Huzzla101, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
We have a big problem... Ancient programmer Pirata dont have a T embed LILYGO CC1101 so he can NOT experiment on it to the fullest extend...

**Describe the solution you'd like**
 I have 40 dollars here now so missing about 30... Out of over 2 thousand people or even more there must be 40 people donating 1 dollar each or even 8 people donating5dollars... So we can send him a LILYGO 

**Describe alternatives you've considered**
 Simple... We have 40 usd with me and when people gather last 40 usd we send him one... Its all about love.. 

**Additional context**
 Think on how much greatness he going to pimp that LILYGO up if he get one as gift to experiment with.... !!! Remember LILYGO is actually on its way to be  harder then the Rock and also it will be GOD POWERS LVL 99 together with ALL  the different platforms that Bruce are on and and off course flipper will integrate to with  us.. A

[image]

That laptop deserve all the dust it has....
As long as we got the right people as Pirata R3ck Pr3y Tawank etc workign on Bruce... THIS WILL BE UNSTOPPABLE.. And Mr Gabriell i see work hard and even more people... Soom the Russian will come then the Asians to .... WORLDWIDE WATERS FOR BRUCE

bmorcelli, 2025-04-05:
Hey man! I really appreciate this campaing, this is trully valid, but I already own a T-Embed CC1101... pr3y and IncursioHack are the ones who doesn't have it.

Supporting Bruce Project (Or a specific dev, like me) is very important, because as we chose to keep the source opened, knowing that some people would use Bruce's codebase or solutions (to create some other payed closed source firmware or whatever).

One of Bruces intentions is explore vulnerabilities and force the companies to develop more secure devices (speaking of RFID, RF and RF technologies), but mostly Bruce wants to allow people to have a few more tools to work with cyber security at a lower price, that's why we are making Bruce as democratic as possible, officially porting it to as many devices as we can.

I know, in my heart, that Bruce needs some enhancements.. which is good since this "new Bruce" is about 11 months old and has evolved without any finantial campaing.. and these enhancements usually come in comparison with Flipper, in matters of:
* RF: Bruce can't decode and work with the miriad of Vendors (some of them rolling codes) that flipper can..
* RFID: Bruce is using PN532 which is very limmited, and Flipper uses other chip which we couldn't find for sale at a good price.
* iButton: Bruce has a simple reader/writer, but can't emulate

Bruce could communicate to Flipper, or be controled by it, but for it I would need to have a Flipper zero, which I don't have, and would be very useful in terms of reverse engeneering (to work with the RF decoders)

So if I could ask for something to enhance Bruces capabilities, it would be a Flipper Zero

Huzzla101, 2025-04-05:
GREAT letter Mr Pirata... Now we got a good overview of the ongoing
situation for Bruce .. Thanks. ok so we will arrange something for
Incursiohack and pr3y and Tawank and R3ck but so you need a flipper?

OK PEOPLE I PUT 40 DOLLAR ON THE TABLE RIGHT NOW, so we need 1 with the
rest or 10 people with a little donation each , this wde can do.. Also we
need to get Pr3y and the other DOPE programmers soemthing...but you have to
understand BRUCE will be big , its already nr 1 on many devices,,... So
this will last for years indeed... KEEP UP THE GREAT WORK ALL YOU SHARKS!!!

Alberto0581, 2025-04-07:
I also donate to the Bruce project. Thank you very much for the work you do

Huzzla101, 2025-04-07:
Your DA CHAMPIONS,..!! i do support monthly some coffe cups but i will also
support 40 usd once i know that people here have donated the other 40usds.
, i dont care how many people we need as long as its gets done...!! And we
are strong enough... I will paste the 40usd coffe receipts here once i know
that you people have donated 40usd minimum together all  as one !! =)
The Sharks need fresh water to so we all can grow!! =)

x01rrr, 2025-04-10:
I think that everyone who is willing to help ,should just donate as much as he want/can 

And Pirata will know how to use that :)

Mawa000, 2025-05-28:
I think I'm a little late but, if possible, I would like to contribute my grain of sand.
How are you organizing the contributions?