BruceDevices/firmware #1043: Bruce starts lagging very much after i opened something in RF

XimKa123, 2025-04-05:
After updating to new firmware Rf worked pretty well but then when i open RF and then spectrum or anything my t embed cc1101 starts lagging so badly that hard reboot required

Huzzla101, 2025-04-05:
your pinout is maybe configured wrong that makes the meny go lagging modus
after you enter RF, have u been in RF config ? put that to cc1101 module if
u using lilygo or put m5 module then go out meny if lags then restart shut
it completely off then start it and should be fine, dont have any external
antennas connected . F.ex if you have correct module but u change Tr or R
pinout it will bug again as it will make a "conflict"

XimKa123, 2025-04-06:
Changing the module in rf to cc1101 worked, thank you!

closed 2025-04-06 by XimKa123