BruceDevices/firmware #1063: RF Spectrum not working

th1s-david, 2025-04-09:
**Describe the bug:**
Attempting to use spectrum function freezes and doesn't display anything 

**Steps To Reproduce:**
Open RF Menu, open spectrum, bring an RF device like a car remote and observe behaviour
-Device used: LilyGO T-Embed CC1101

**Expected behavior:**
Visible waves

**Actual behavior:**
Sadness

**Additional context / logs:**
https://drive.google.com/file/d/1QJaYS7yE5VeaWScQtXx4tvXXNobT1Ifb/view?usp=drivesdk

Used to work before, and reading with read tag works normally so it's an issue with the spectrum itself.

th1s-david, 2025-04-09:
Ignore my awesome socks please and thank you.

Alema1, 2025-04-10:
@th1s-david Does you T-Embed capture the signal from you key using the scan/copy? I've tested on version 1.7 and works, 1.9.1 won't capture anything here.

rennancockles, 2025-04-10:
RF spectrum has a bug, but we're already working on it

th1s-david, 2025-04-10:
love y'all the effort is insane thanks ❤️

rennancockles, 2025-04-12:
Fixed in Beta

closed 2025-04-12 by rennancockles