BruceDevices/firmware #1073: Add led-blinking on T-Embed when doing something

TurboKoT1, 2025-04-12:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like the bottom left LED on the T-Embed to flash a certain color like in flipper zero when you for example:
- Use IR: TV-B-Gone - LED blinks blue for each successfully sent command; successfully recorded - glows green
- Using RF: Send/Record - blinks blue; Successfully recorded - glows green.
- Charging the device (lights red)

**Describe alternatives you've considered**

**Additional context**
[image]
i mean this led

th1s-david, 2025-04-12:
Would be cooler if you could adjust which led it is or you could set it to all, and also if you could set what color is for what action

bmorcelli, 2025-05-26:
led blinks when capturing RF signals