BruceDevices/firmware #1074: Attempting to select SubGHZ from RF returns to main menu

th1s-david, 2025-04-12:
**Describe the bug:**
Unable to choose a SubGhz file from RF Custom SubGhz menu

**Steps To Reproduce:**
Go to RF -> Custom SubGhz -> Choose LittleFS
- Device used:  LILYGO T-Embed CC1101

**Expected behavior:**
Opens LittleFS to choose a file

**Actual behavior:**
Returns to main menu.

**Additional context / logs:**
Bruce Version: 60c4d95

You can use subghz files if you go to LittleFS manually and select the files and press the transmit thing

https://files.catbox.moe/gayjur.mp4

rennancockles, 2025-04-12:
We're already working on a fix. In the meantime you can unblock your usage creating a BruceRF folder into LittleFS and moving your .sub files there

rennancockles, 2025-04-12:
fixed in beta

closed 2025-04-12 by rennancockles