BruceDevices/firmware #1075: buffering .sub files before playing them

Gizzzgg, 2025-04-12:
**Problem**:

combined module with cc1101 and NRF24 in one on a sd card sniffer, but it's without sd card slot:

[image]

no possibility to play subghz files from sd card using this module.

**solution**:

1. user selects sub file from sd card

2. file buffers

3. user swaps sd card to module

4. user presses OK, sub file starts playing.

**additionally**:
add pause after file played, so user could swap to sd card back.

Also, i know that i can add files to littleFS, but it's only 512kb (even less with custom boot logo), which is not enough to load whole subghz database.

PopperVids, 2025-04-14:
What is the module called

Gizzzgg, 2025-04-14:
> What is the module called 

literally "cc1101 & NRF24", but it's not available anywhere beside Russia, at least for now

bmorcelli, 2025-04-14:
> but it's without sd card slot

How would you read from SD Card if there isn't an SD Card attached?

LittleFS is a small piece of flash memory to use for this purpose, so you need to choose well what you'll do with it.

If you flash Bruce without the Launcher, you will have almost 3mb of memory in the LittleFS

Gizzzgg, 2025-04-14:
> How would you read from SD Card if there isn't an SD Card attached?
>
> LittleFS is a small piece of flash memory to use for this purpose, so you need to choose well what you'll do with it.
>
> If you [...]

my idea is to connect SD card, select .sub file from it as usual, but instead of playing it right away, load file to littleFS and wait until user press enter.

while it's loaded to littleFS you can swap from SD card to this module and play it after pressing enter

(i probably explaining really bad, sorry, English is not my first language)

rennancockles, 2025-04-14:
I'm sorry but I don't think it's worth it. It's too much to change and it would make the general usability worse just to make it work for your special case. You can do it by yourself though and copy the file you want to little fs before using it in custom subghz.

closed 2025-04-23 by rennancockles