BruceDevices/firmware #1079: Brucegotchi stops channel hopping

psionyx2311, 2025-04-12:
**Describe the bug:**
After 1-2 handshake catches, an "Error:Unknown" begins to cycle amongst the messages. After that is fitst seen, channel hopping freezes on whatever channel that error message occured on.
Always after a 1-2 handshake catchs. If 0 handshakes, channel hopping cycles fine, no error messages.
This has been tried on multiple 2.8 CYD, same symptom, reproduceable.

**Steps To Reproduce:**
1) Start Brucegotchi
2) wait until 1-2 handshakes occur
3) watch for "Error:Unknown"
4) see channel hopping freeze
5) periodic error unknown display
- Device used: 
ESP32-2432S028
**Expected behavior:**
Channel hopping reguardless of handshake capture. Not being on 1 channel until restarted.

**Actual behavior:**
Channel hop freezes after seeing the first Error:Unknown, everytime.

**Additional context / logs:**
Picture of the error, and being on Channel 1 for near 40mins

[image]

Krutifulmen, 2025-05-25:
Same, but on m5 stick with 0 handshakes cathes

bmorcelli, 2025-05-27:
It is happening due to a Low memory error.. I tried to fix it today and hope it is working fine now...

You can test it on Beta

closed 2025-05-27 by bmorcelli