BruceDevices/firmware #1080: Bruce fails to read SD card

PopperVids, 2025-04-13:
**Describe the bug:**
Bruce fails to read any SD card

**Steps To Reproduce:**
Go to files
- Device used: ESP32-2432S028

**Expected behavior:**
Lets me view files on the SD card

**Actual behavior:**
Work sometimes for the first page but when you press view on a file its blank and if you go back you start getting weird symbols in the top left

**Additional context / logs:**
I am using the launcher and it has no issues reading the SD card the SD card is 16GB and formatted to FAT32 Bruce version 1.9.1

[image]

maxim1470, 2025-04-26:
me have same problem its very bad bug

bmorcelli, 2025-04-26:
are you sharing this SD Card with other bruce devices??

if so:
* Take the SD Out, go Files> LittleFS> bruce.conf> delete
* restart your device
* Put the SD Card back in, Files> SD Card> bruce.conf> delete

the pin definitions, at the moment, are stored in the bruce.conf file.. so if you use it in multiple bruce devices it will fail becaus the pins will be different within the different devices..

we are working to make it "device related" and being able to use the same SD Card with many bruce devices without a problem

maxim1470, 2025-04-26:
the sd is clear  but it dont see files of cyd